AllenCellModeling/aicsimageio #4: CZI dimensions not being read properly for multi scene

JacksonMaxfield, 2019-06-27:
Reproducible using the largest czi in the resources directory:

```python
from aicsimageio import AICSImage

im = AICSImage("s_3_t_1_c_3_z_5.czi")
im.data.shape  # returns (3, 1, 3, 5, 4029, 5476)
```

It finds the correct `Scene` size but something is wonky about the `Y` and `X`. `Y` should be 325 and `X` should be 475.

For comparison here is the output of reading the smallest czi:
```python
im = AICSImage("s_1_t_1_c_1_z_1.czi")
im.data.shape  # returns (1, 1, 1, 1, 325, 475)
```

JacksonMaxfield, 2019-09-10:
Long delayed update: This is being fixed by upgrading `pylibczi` and using it as our CZI reader thanks to @heeler. That update will hopefully be pulled in soon^tm.

heeler, 2019-09-10:
So what I think is happening here is that the current reader for czi's uses CziFile to read the image. I don't believe it supports multi-scene so the fact that it reads at all supprises me. Once we implement the new reader we should get back an appropriately shaped Numpy.NDArray.

PR #63 (2020-02-03): feature/use-dask
**Pull request recommendations:**
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-12], adds tiff file format support_
Resolves #4 
Resolves #45 
Resolves #53 
Resolves #56 
Resolves #57 
Resolves #58 
Resolves #59 
Resolves #60 
Resolves #67 
Resolves #68 
Resolves #69 

- [x] Provide context of changes.
Dask is pretty cool and we are starting to lean on it pretty heavily. Specifically in the case of multi-hundred-GB to TB large CZI files, we needed a [... cut]
closed 2020-02-03 by JacksonMaxfield