AllenCellModeling/aicsimageio #18: Fix dim order for `OmeTIFFWriter`

JacksonMaxfield, 2019-09-10:
The `writers` module has been updated with docstrings but hasn't been updated to follow our new spec. Currently it saves the data out in `TZCYX` instead of our new `STCZYX`.

toloudis, 2019-09-10:
Why not pass in the dim order as a string?  Also, multi-scene ome-tiff is a bit more complicated because you have to create multiple <Image>s in the xml.  I would consider splitting this into 2 issues, one for arbitrary (T,C,Z)YX dimension order, and then another for multi-scene output.

JacksonMaxfield, 2019-09-10:
Agree. Will retain this one as the Arbitrary dim order issue. By default it should be `TCZYX` order yes?

toloudis, 2019-09-10:
Yes I think TCZYX is the most natural default for our data even though it is an interface change.

toloudis, 2019-10-28:
I think #32 resolves this now.

toloudis, 2019-10-28:
Except for the default dimension ordering, which was left as TZCYX for now, so as not to break client code immediately.

closed 2019-10-28 by JacksonMaxfield