AllenCellModeling/aicsimageio #55: Channel & Z Switched?

TeddyZ95, 2020-01-13:


JacksonMaxfield, 2020-01-13:
Hey @TeddyZ95

Need a bit more context. Could you copy paste the code that is breaking as a comment here? And perhaps you could also send the file that is breaking.

TeddyZ95, 2020-01-13:
Download full resolution images
Available until Feb 12, 2020
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAXVK6zxVlYJKceab76D1r0zs4wmqAe5kyoyOlQ8VC44EbTMrLQjSy78s%2F%24%7Bf%7D%3Fo%3DAoeYPcE6OwiUtY8prx2WYvbREi5sP9Pgd32y4upmds6_%26v%3D1%26x%3D3%26a%3DCAognszbUPQ9_Hfp3VhwQ7Zi2cdF5tnY_FFnhGsFXcU75h4SJxD43aeI-i0Y-O2i3IMuIgEAKggByAD_ER2KHFIE7OMJqloE0su_LA%26e%3D1581546780%26k%3D%24%7Buk%7D%26fl%3D%26r%3D480F8AE0-3AF9-429D-83D0-BDA34BF276B2-1%26ckc%3Dcom.apple.largeattachment%26ckz%3D4CC3C6BA-7B7A-47F7-BBFE-5C31C285A73C%26p%3D19%26s%3D0phT0yU-etQMCrSrXMDple4-tx0&uk=xU4RRja0WOr1Mz1DRTVf0g&f=Images.zip&sz=33558125>I have a tiff file that has 4 channels and only 1 z, t, and s dimensions. I’ll attach the file to this email 

However, when I load the file using aicsimage and look at the np.shape it returns (1, 1, 1, 4, 2048,2048) which leads me to believe that the ordering if mixed up. Since in the documentation says its suppose to come back as STCZYX.

Best, 

Ted

JacksonMaxfield, 2020-01-13:
Thanks for sending over the image! What is happening is that because there is no metadata for the image, our tiff reader is guessing the dimension order. For Tiffs we assume the dimension order is "AICS normal" ordered, which for us means: in `STCZYX` order. In this case though, you know the dimension order and our guess is wrong so the solution would be to use the `known_dims` keyword argument.

```python
from aicsimageio import AICSImage

with AICSImage("/home/maxfield/Downloads/883_G11_1_all.tif") as img:
    print(img.data.shape)  # returns (1, 1, 1, 4, 2048, 2048)
    print(img.dims)  # returns 'STCZYX'

with AICSImage("/home/maxfield/Downloads/883_G11_1_all.tif", known_dims="CYX") as img:
    print(img.data.shape)  # returns (1, 1, 4, 1, 2048, 2048)
    print(img.dims)  # returns 'STCZYX'
```

We should make this more obvious that we are guessing dimension order when there is no metadata present in the file and we will put in a patch to at least log a warning to inform you in the future.

TeddyZ95, 2020-01-13:
Right. But I was also wondering say I don’t know the dims either would there be any way to validate that say the returned dimensions are what they are? 

Sorry if that sentence is confusing. 

Ted

JacksonMaxfield, 2020-01-13:
Without metadata the only way in my head of "validating" that one dimension is channels over some other dimension is to turn the channels dimension into RGBA and save the image as a PNG and see if it looks right? I guess what I am saying is that without metadata the best way to validate that a dimension is a dimension is human guessing.

closed 2020-01-14 by JacksonMaxfield