AllenCellModeling/aicsimageio #69: get_image_dask does not handle chunk_by_dims

vianamp, 2020-01-30:
This code

```
if __name__ == "__main__":
	with AICSImage("20190809_R02.czi", chunk_by_dims=['Y','X']) as raw:
		img = raw.get_image_dask_data("TYX", S=0, C=0, Z=12)
		print(img.shape)
```

gives this error:

```
Process Dask Worker process (from Nanny):
Traceback (most recent call last):
  File "/home/matheus.viana/anaconda3/envs/aicsimageio-test/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/matheus.viana/anaconda3/envs/aicsimageio-test/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/matheus.viana/anaconda3/envs/aicsimageio-test/lib/python3.7/site-packages/distributed/process.py", line 191, in _run
    target(*args, **kwargs)
  File "/home/matheus.viana/anaconda3/envs/aicsimageio-test/lib/python3.7/site-packages/distributed/nanny.py", line 674, in _run
    worker = Worker(**worker_kwargs)
  File "/home/matheus.viana/anaconda3/envs/aicsimageio-test/lib/python3.7/site-packages/distributed/worker.py", line 638, in __init__
    **kwargs
TypeError: __init__() got an unexpected keyword argument 'chunk_by_dims
```

JacksonMaxfield, 2020-01-31:
Valid bug find for v3.1.0

This is because I am naively passing all kwargs from the AICSImage object down to Dask. Two potential solutions:

1. make a specific `dask_kwargs` kwarg for AICSImage that whatever you pass as a dictionary to it will be passed down to dask.
2. expose all dask kwargs as parameters and specifically pass them all down.

Will discuss with others as to which one to implement. None would be breaking changes to what you pasted above.

JacksonMaxfield, 2020-01-31:
@vianamp A patch for this has been pushed to the branch. The solution we choose was basically compartmentalizing dask kwargs to their own parameter.

You above code snippet will work with the changes but in the future, if you want to specify dask kwargs you will do the following:

```python
with AICSImage("my_image.tiff", dask_kwargs={"nworkers": 4}) as img:
    # my processing
```

commit 0b11a3ef (2020-01-31): Resolves #69, Change context manager API to use dask_kwargs param
PR #63 (2020-02-03): feature/use-dask
**Pull request recommendations:**
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-12], adds tiff file format support_
Resolves #4 
Resolves #45 
Resolves #53 
Resolves #56 
Resolves #57 
Resolves #58 
Resolves #59 
Resolves #60 
Resolves #67 
Resolves #68 
Resolves #69 

- [x] Provide context of changes.
Dask is pretty cool and we are starting to lean on it pretty heavily. Specifically in the case of multi-hundred-GB to TB large CZI files, we needed a [... cut]
closed 2020-02-03 by JacksonMaxfield