AllenCellModeling/aicsimageio #78: get_physical_pixel_size should be a function on AICSImage

toloudis, 2020-03-05:
## Use Case
AICSImage supports get_channel_names, but you have to get the reader to call get_physical_pixel_size

## Solution
Make get_physical_pixel_size behave the same way as get_channel_names

JacksonMaxfield, 2020-03-06:
I can take this one but would like to know what you think the default should be in the case where metadata for the image isn't available? A tuple of `None` for each dimension?

`get_channel_names` returns a list of ints for each dimension if no metadata is available but this is a harder case for default values imo.

AetherUnbound, 2020-03-06:
Agreed - what would be the default for PNGs, for example?

toloudis, 2020-03-06:
For PNGs it would be [1,1,1].  Default values are 1,1,1 for everything unless discovered in the metadata.

toloudis, 2020-03-06:
To be more specific: The only physical spatial dimensions are x,y,z.  And defaults are [1,1,1] returned in whatever format the existing readers return them in.

JacksonMaxfield, 2020-03-06:
Ahhh yep. Only spatial dimensions need to be returned. The tricky thing here is that DPI is sometimes provided in the metadata of files from `DefaultReader` for example:

```python
In [4]: img = AICSImage("aicsimageio/tests/resources/example.jpg")                                                                                                                                                   

In [5]: img.metadata                                                                                                                                                                                                 
Out[5]: 
{'exif': b'Exif\x00\x00II*\x00\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
 'dpi': (72, 72),
 'adobe': 100,
 'adobe_transform': 100}
```

Couldn't you simply invert that value to get the physical pixel size for `DefaultReader` images? (If available)

toloudis, 2020-03-06:
I suppose that would work.  One problem is that we are not careful about units.  The units can be wildly inconsistent.  Microns, inches, meters, whatever.  It's likely that most people reading 2D PNG or JPG images using this library will not care one bit about DPI.   But I suppose it is more "future-proof" if you can get the DPI from the image.

toloudis, 2020-03-06:
Bear in mind that the DPI of a 2D image is not the same as the physical pixel size of a volume image.  It has a very different meaning and we'd be sort of overloading the meaning of physical_pixel_size.

JacksonMaxfield, 2020-03-06:
Fair. I think in this case I would say, if we can find the metadata in the image for physical pixel size, then great return it as a list, otherwise return a list of `[1, 1, 1]` and issue a warning in the log.

JacksonMaxfield, 2020-03-06:
@toloudis, currently `OmeTiffReader` has this function exposed but returns them in order `X`, `Y`, `Z`.

Should this be changed to `Z`, `Y`, `X` to stay consistent with the rest of `aicsimageio`?

PR #80 (2020-03-06): feature/get-physical-pixel-size
**Pull request recommendations:**
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-12], adds tiff file format support_

Resolves #78 

- [x] Provide context of changes.

Dan wanted something, so I added it.

For real though, this implements the same getter that `get_channel_names` uses to go to the base reader for pixel sizes and return. My one gripe with this is that the values are returned in `X`, `Y`, `Z` order which is the opposite o [... cut]
closed 2020-03-06 by JacksonMaxfield