AllenCellModeling/aicsimageio #79: pixels_physical_size in omeTifWriter not applying to image

mdylan2, 2020-03-06:
## System and Software
* aicsimageio Version: Latest - downloaded today
* Python Version: 3.6
* Operating System: Windows

## Description
I am trying to save a segmentation from my Jupyter Notebook to my computer as TIFF. Below is the code that I am trying to use:
[image]

## Expected Behavior
I expected that the saved image would have voxel spacing as I specified above. After opening the image in ImageJ, this is not the case. Image info from ImageJ attached below:
[image]

Am I doing anything wrong here?

toloudis, 2020-03-06:
My first suspicion is that if you load in ImageJ using File-->Import-->BioFormats, then I will feel more certain that the voxel size is being read properly from the OME XML metadata.  If you are already doing that, then possibly renaming to .ome.tiff will be a better hint that this file contains OME XML metadata.  Failing those, then we will look deeper and see if we have to fix something.

mdylan2, 2020-03-06:
You're right. This issue is resolved. Solution [here](https://forum.allencell.org/t/writing-to-a-tiff-file-with-voxel-spacing-error-in-codebase/184/5).

Thanks for the help guys!

closed 2020-03-06 by mdylan2