AllenCellModeling/aicsimageio #82: Can't import package without network access

mruffalo, 2020-03-11:
## System and Software
* aicsimageio Version: 3.1.2
* Python Version: 3.8.1
* Operating System: Docker container built from `python:3` base image

## Description
Without network access, the `aicsimageio` package cannot be imported.

We're trying to use this as part of a pipeline written in CWL, with each tool encapsulated in a Docker container. By default, the `cwltool` command-line workflow runner invokes Docker with `--net=none`, which causes imports of `aicsimageio` to fail.

This is straightforward for us to fix/work around, but it seems like this package should probably be usable without network access.

## Expected Behavior
`import aicsimageio` should succeed without network access.

## Reproduction

```
$ docker run -it --net=none hubmap/codex-scripts
Python 3.8.1 (default, Feb  2 2020, 08:37:37) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import aicsimageio
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/toolz/functoolz.py", line 456, in memof
    return cache[k]
KeyError: ('8.8.8.8', 80)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/distributed/utils.py", line 138, in _get_ip
    sock.connect((host, port))
OSError: [Errno 101] Network is unreachable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/site-packages/aicsimageio/__init__.py", line 4, in <module>
    from .aics_image import AICSImage  # noqa: F401
  File "/usr/local/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 9, in <module>
    from distributed import Client, LocalCluster
  File "/usr/local/lib/python3.8/site-packages/distributed/__init__.py", line 3, in <module>
    from .actor import Actor, ActorFuture
  File "/usr/local/lib/python3.8/site-packages/distributed/actor.py", line 6, in <module>
    from .client import Future, default_client
  File "/usr/local/lib/python3.8/site-packages/distributed/client.py", line 44, in <module>
    from .batched import BatchedSend
  File "/usr/local/lib/python3.8/site-packages/distributed/batched.py", line 8, in <module>
    from .core import CommClosedError
  File "/usr/local/lib/python3.8/site-packages/distributed/core.py", line 17, in <module>
    from .comm import (
  File "/usr/local/lib/python3.8/site-packages/distributed/comm/__init__.py", line 25, in <module>
    _register_transports()
  File "/usr/local/lib/python3.8/site-packages/distributed/comm/__init__.py", line 16, in _register_transports
    from . import inproc
  File "/usr/local/lib/python3.8/site-packages/distributed/comm/inproc.py", line 75, in <module>
    global_manager = Manager()
  File "/usr/local/lib/python3.8/site-packages/distributed/comm/inproc.py", line 39, in __init__
    self.ip = get_ip()
  File "/usr/local/lib/python3.8/site-packages/distributed/utils.py", line 162, in get_ip
    return _get_ip(host, port, family=socket.AF_INET)
  File "/usr/local/lib/python3.8/site-packages/toolz/functoolz.py", line 460, in memof
    cache[k] = result = func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/distributed/utils.py", line 147, in _get_ip
    addr_info = socket.getaddrinfo(
  File "/usr/local/lib/python3.8/socket.py", line 918, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Temporary failure in name resolution
```

## Environment
N/A

JacksonMaxfield, 2020-03-11:
First of all, thanks for alerting us.

It seems that one our dependencies which allows for parallel image reading is causing the issue.
Specifically, it isn't that it needs to be connected to an external network, it that it is trying to spawn ports for worker cross communication.

Regardless, we will try to figure out a fix.

JacksonMaxfield, 2020-03-11:
@mruffalo Just merged a PR that should hopefully solve this. (At least I tested it with a `python:3` docker image with `--net=none` flag. Will release a new version of the package soon and hopefully you can upgrade and it will work just fine.

closed 2020-03-11 by JacksonMaxfield

JacksonMaxfield, 2020-03-11:
@mruffalo `aicsimageio==3.1.3` has been released. It should fix this issue :)

mruffalo, 2020-03-13:
@JacksonMaxfield Wow, that was fast. Much appreciated! This was trivial for us to work around: https://github.com/hubmapconsortium/codex-pipeline/commit/3749966585421c5699725bcf55450236558e7db1, but this still seemed worth reporting. Thanks!

PR #83 (2020-03-11): bugfix/delay-import-of-distributed-module
**Pull request recommendations:**
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-12], adds tiff file format support_

Resolves #82 

- [x] Provide context of changes.

During the import of `aicsimageio`, `distributed` was also being imported which somewhere along it's import chain, is a socket test to ensure that it can spawn cross communication ports between workers.

This is at odds with running processes on machines or in  [... cut]