AllenCellModeling/aicsimageio #86: TiffReader and DefaultReader do not have `get_channel_names` functions

JacksonMaxfield, 2020-03-19:
Somewhat standardizing with the `get_physical_pixel_size`, all readers should implement or inherit from the base class `get_channel_names` function.

toloudis, 2020-03-19:
Unless python inheritance works differently than I expect, if the base class reader supplies an implementation of get_channel_names, then the derived class readers don't need to do anything else.  They should already be inheriting it, right?

JacksonMaxfield, 2020-03-19:
Correct! Which is the implementation I will be using for this, just hasn't be done yet.

toloudis, 2020-03-19:
I see now, I missed that there is no Base reader's get_channel_names.  The length of the list returned should match the number of channels, so it might be tricky.  What would the default be?  Empty list? One element?

PR #88 (2020-03-20): feature/add-get-channel-names-to-base-reader
**Pull request recommendations:**
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-12], adds tiff file format support_

Resolves #86 

- [x] Provide context of changes.

Follow the spec from `get_physical_pixel_size` to expose the function on the base reader then mutate on inheriting readers for `get_channel_names`. All readers will now at a minimum return a `List[str]` for `n` channels they contain or `None` when no channel [... cut]
closed 2020-03-20 by JacksonMaxfield