AllenCellModeling/aicsimageio #101: Write generalized "Given some reader function" produce dask array

JacksonMaxfield, 2020-04-06:
After reviewing #99 and seeing that it's `_daread` is an exact copy of `CZIReader._daread`, it would be useful to expose a `_daread` function on the base reader so that Reader authors only have to write the `_imread` function as long as it conforms to spec. This is an entirely optional implementation. The authors can still implement their own readers however they want but so far there are two readers with the same code and I suspect it will only grow as we address #89

toloudis, 2020-09-14:
My translation: see if there is a DRY refactor that is generally applicable to all readers trying to do this

JacksonMaxfield, 2021-02-20:
Closing this because it was created during the 3.x work and in creating the 4.x readers I have realized we don't need this.

closed 2021-02-20 by JacksonMaxfield