AllenCellModeling/aicsimageio #103: Allow passing `range` or a sequence of integers to image data getters

JacksonMaxfield, 2020-04-09:
## Use Case
*Please provide a use case to help us understand your request in context*
Currently when requesting chunk or slice data from `AICSImage.get_image_data` and `dask_data`, the user can only specify a single integer for the dimensions not explicitly specified I.E.

```python
AICSImage.get_image_data("ZYX", S=0, T=0, C=0)
```

That is the limit of the request / parameter format. However, there is an obvious use case, esp with timeseries data for `range` requests.

```python
AICSImage.get_image_data("TZYX", S=0, T=range(0, 100, 5) C=0)
```
This means, gather every fifth timepoint and merge into a single array.

## Solution
*Please describe your ideal solution*
The above use case gives some examples but here are more:

Get the first and second channels
```python
AICSImage.get_image_data("CZYX", S=0, T=0, C=range(0, 2))
```

Get the first and second channels using a `Tuple[int]`
```python
AICSImage.get_image_data("CZYX", S=0, T=0, C=(0, 2))
```

Get the first and second channels using a list of indices
```python
AICSImage.get_image_data("CZYX", S=0, T=0, C=[0, 1])
```

The last one is an idea as I believe there may be value in it when you combine metadata:
```python
AICSImage.get_image_data(
    "CZYX",
    S=0,
    T=0,
    C=[channels.index(c) for c in channels if c in ["EGFP", "RFP"]]
)
```

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*
Using normal `dask` and `numpy` array slicing:

```python
img = AICSImage("my_file.tiff")
lazy_chunk = img.get_image_dask_data("ZYX", S=0, T=0, C=0)
middle_slice = lazy_chunk[20:30, :].compute()
```

It's _fine_ and I wouldn't discourage people from doing it, but I don't see why adding the range / list of index functionality is bad either.

toloudis, 2020-04-09:
the full python ndarray slicing approach also lets you specify XY subsets as well.  Can we do that with `X=range(65,75)` ?  I think it would be ok even if readers had to read the whole planes and then slice after the fact.  An example use case of this would be loading a single segmented cell from a volume file!

JacksonMaxfield, 2020-04-09:
We could allow `YX` dims to be passed in the get image data but there is little value in doing that other than doing the operation for the user. (Which is basically this entire github issue but still), since the base dask delayed reader is a `YX` plane, the best we can do is read the plane, then slice. Whereas the rest of this github issue is specifying which planes to read.

toloudis, 2020-04-09:
There's another possible entry which is allowing slice objects as well.  `T=slice(start, stop, step)` which then opens the door to full slicing syntax.  ( https://stackoverflow.com/questions/509211/understanding-slice-notation )

JacksonMaxfield, 2020-04-09:
Ahh yes, in my above examples `range` and `slice` should be interoperable.

wiegraebe, 2020-04-15:
My questions are driven by the following use cases:
1) z-stacks have often several slices without useful information (e.g. coverslip under the cell and buffer above it). For initial analysis or visualization it is often sufficient to load only slices that most likely have information about the cell. This would require to load all slices between an upper and lower limit.
2) For initial analysis (e.g. to find the z-range that contains a cell, judge quality of image), it might be useful to read only a sample of slices (e.g. every 5th slice).
3) Microscope images have often edge artifacts in xy (e.g. processing artifacts from filters and deconvolution, uneven illumination). Therefor we often reject e.g. 10% of the border rows and columns.

An extension of the approach discussed so far would be the introduction of arbitrary mask. Use cases would be:
1) We know from one channel (e.g. transmitted light) that part of the image contains no relevant information (e.g. background, no colonies, dead cells). We want to increase loading speed of the other channels (e.g. membrane, nucleus, eGFP structure) by applying a mask before loading.
2) The Airy fast detector has 32 channels. We know what channel has the brightest information. We could mask the other channels before loading areas that are overexposed in the brightest channel.

toloudis, 2020-04-16:
That's great information, and it sounds like the proposed interface would handle everything.   Masking would be a completely different interface that we'd have to design separately and I'm not sure it would add much that you can't do by just choosing ranges to load, with a little python math.

@JacksonMaxfield will it raise an exception if the requested ranges don't intersect the actual dimensions?

JacksonMaxfield, 2020-04-16:
Seconding the "this is great information".

> @JacksonMaxfield will it raise an exception if the requested ranges don't intersect the actual dimensions?

That is the behavior I would want yes.

> An extension of the approach discussed so far would be the introduction of arbitrary mask...
I think I agree with @toloudis that this is a separate interface potentially. But can't even think of the psuedo-code that would get us there.

PR #109 (2020-04-29): feature/allow-sequence-in-get-data
## Description
`transforms.reshape_data` (and by extension, `AICSImage.get_image_dask_data` and `AICSImage.get_image_data`) now support dimension kwargs being handed `int`, `List[int]`, `Tuple[int]`, `range`, and `slice`.

I pretty much rewrote the current `reshape_data` function to be much more generalized and rely less on `numpy` and `dask` functions and rather just array operations based off dimension ordering and selection. A lot of these types of operations are found throughout the repo already with `getitem_ops` or `ops` in general.

## Pull request rec [... cut]
commit 9ce6439c (2020-04-29): feature/allow-sequence-in-get-data (#109)

* Resolves #103, allow range, list, tuple, slice to reshape data

* All prior tests passing, bugfixes to empty dimension insertion

* Black formatting

* Add basic tests

* Remove error raising on slice syntax

* Add tests for deep value checks with kwargs and reorder

* Update documentation

* Fix black and flake8 errors

* Add missing ConflictArgumentsError test

* Changes from initial feedback

* Add more documentation

* Slices can now IndexError if the stop index is greater than dim size

* Add negative value checking to index errors

* Use carte [... cut]
closed 2020-04-29 by JacksonMaxfield