AllenCellModeling/aicsimageio #104: Detect No Cluster and read single shot

JacksonMaxfield, 2020-04-14:
## Use Case
*Please provide a use case to help us understand your request in context*
[image]

Make the blue dots be comparable to the the orange dots, on "No Cluster"

Speed up image reading when no cluster.

## Solution
*Please describe your ideal solution*
Do the image read with the base libraries but still read into a dask array.

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*

JacksonMaxfield, 2020-04-15:
Must also detect memory available prior to reading in the image in full. If the image cannot fit in memory, log a warning to the user that explains that the image will be read in a delayed fashion, if they want to speed up this process, the should spawn a local cluster.

toloudis, 2020-04-15:
I don't know if detecting available memory is worthwhile.  Memory conditions can change from moment to moment.  Other processes could be starting or ending.   Standard out-of-memory exceptions are a very natural way to handle it.  Users should chunk their reads if they are worried about memory conditions.

JacksonMaxfield, 2020-04-15:
Yea on further thinking, the user should know when the image is too large or not.

PR #113 (2020-05-06): feature/optimize-readers
## Description
Optimizing our single-threaded read performance. Because we want to optimize all readers at once and not have each reader optimize itself, this results in somewhat of a change to the reader spec. Instead of readers implementing `dask_data` alone, readers now need to implement `_build_delayed_dask_data` and `_read_in_memory_data` which the base reader will determine which to use based off of a `Distributed.Client` being available to the process or not.

New read times:
[image]

## Pull request recommendations:
- [x] Name your pull request _your-developmen [... cut]
closed 2020-05-06 by JacksonMaxfield