AllenCellModeling/aicsimageio #105: Write benchmark suite and integrate into CI

JacksonMaxfield, 2020-04-15:
## Use Case
*Please provide a use case to help us understand your request in context*
Related to #104 

Monitor image read performance on every commit to master to watch for performance hits introduced with each patch.

## Solution
*Please describe your ideal solution*
Look into [asv](https://asv.readthedocs.io/en/stable/writing_benchmarks.html) which is used by pandas for benchmarking which is generally a good sign.

On every commit to master, run the benchmarks with different configurations on fargate instances? Potentially?

GitHub Actions server replicates no cluster
Fargate single node 8 vCPU replicates standard local-cluster
Fargate single node 32 vCPU replicates nice local-cluster
Fargate multi node 32 * 1 vCPU replicates standard distributed cluster
Fargate multi node 100 * 1 vCPU replicates nice distributed cluster

Each benchmark suite publishes image and CSV to S3 quilt package to version benchmarks

Single gather job at the end to pull all individual CSVs together and create final benchmark image, upload to S3 at standard URL so that the image can be displayed on README.

This would require building and publishing a Docker image prior to running the fargate clusters.

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*
Use the janky script currently written with a few patches.

toloudis, 2020-04-15:
benchmarks are a cool idea.  Does it cause too many extra dependencies?  Maybe it could be a separate repo: aicsimageio-benchmarks.  Also, Fargate = $.  Maybe it should only run automatically on more strict conditions, like tagged releases?

JacksonMaxfield, 2020-04-15:
I would probably just put the extra dependencies in a `.[benchmarks]` category so that they aren't installed unless the CI or we are running them. Correct on the $$$, and yea I think this should run on just pushes to `stable` which is basically the equivalent of pushing on tagged release.

toloudis, 2020-09-14:
currently done manually:  
```
python scripts/benchmark.py
python scripts/chart_benchmarks.py
```

closed 2021-01-21 by JacksonMaxfield