AllenCellModeling/aicsimageio #106: Allow `AbstractBufferedFile` as input to AICSImage and Readers

JacksonMaxfield, 2020-04-17:
## Use Case
*Please provide a use case to help us understand your request in context*
If I want to read a file from an S3, GCS, or Azure bucket, I would currently first need to download the file locally then read it. Not optimal in two capacities.

1. If the user doesn't have enough storage on the local machine in the first place
2. If the compute is being done by EC2 or GC Compute Engine, or etc, then there would be very little value in copying to the worker.

## Solution
*Please describe your ideal solution*

`AbstractBufferedFile` is the ABC from `fsspec` which is the basis for the file system handlers for S3, GCS, Azure, etc (`s3fs`) for example. By allowing all children of `AbstractBufferedFile` we are allowing all `S3File` and the like. We are not returning to open buffer handling. `S3File` is similar to `Path` in a sense where it is just a pointer to something, and notably, `S3File` and `AbstractBufferedFile` in general can be pickled, meaning distributed reads to these targets still work. The underlying actual reading libraries (`tifffile`, `aicspylibczi`, `readlif`, etc.) simply need to support reading from an open buffer.

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*
As mentioned above, download the file in full then read.

toloudis, 2020-04-17:
Is pickleable a documented requirement for all derived classes from AbstractBufferedFile ?

JacksonMaxfield, 2020-04-17:
> Is pickleable a documented requirement for all derived classes from AbstractBufferedFile ?

Great question. [Serialization is a part of the spec](https://filesystem-spec.readthedocs.io/en/latest/features.html?highlight=pickle#serialisability)

JacksonMaxfield, 2020-05-28:
Update to this: In implementing, configure read routing to be "Remote or local" as for some file formats this is really slow.

JacksonMaxfield, 2020-09-11:
Maybe just allow string uri: https://github.com/intake/filesystem_spec/blob/master/fsspec/core.py#L534

closed 2020-12-14 by JacksonMaxfield