AllenCellModeling/aicsimageio #110: Add `transform_metadata` function to CziReader and AICSImage

JacksonMaxfield, 2020-04-29:
## Use Case
*Please provide a use case to help us understand your request in context*
The CZI to OME metadata XSLT project is close to beta testing. Adding a function to the readers (`transform_metadata`) that transforms their current metadata to OME-4DN would be great to having a unified API for metadata handling.

## Solution
*Please describe your ideal solution*
`CziReader.transform_metadata() -> ElementTree`
`Reader.transform_metadata(): NotImplementedError()`
`AICSImage.transform_metadata(): self.reader.transform_metadata()`

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*

toloudis, 2020-04-29:
Alternatives:
get_ome_metadata
get_metadata_as_ome
transform_metadata_to_ome

JacksonMaxfield, 2021-06-14:
@MatteBailey this is the goal.

We need to submodule into a new "metadata" module. See also #111 

```python
from aicsimageio import AICSImage
from aicsimageio.metadata import transform_czi

# in CziReader
# @property
# def ome_metadata(self) -> OME:
#     return metadata.transform_czi(self.metadata)

img = AICSImage("my-file.czi")
transformed = img.ome_metadata
img.save("converted.ome.tiff", ome=transformed)
# img.save("converted.ome.tiff", transform_metadata=True)
```

PR #289 (2021-07-16): feature/ome-metadata-xslt-spec
## Description

Adds generic transformation function that we can reuse for all readers as XSLT submodules come in as well as defines the Reader and AICSImage spec for `ome_metadata` property.

Sets up @MatteBailey nicely for his PR which will add CZI-to-OME metadata transformation.

I have already tried the function manually with just a path to a local directory, not with a submodule or anything and everything "works" but am getting validation errors that I know @MatteBailey said he was in the process of wrapping up.

I also need to add in a future PR, an update  [... cut]
closed 2021-07-16 by JacksonMaxfield