AllenCellModeling/aicsimageio #115: Update OME-XML spec to the latest version

sebi06, 2020-05-06:
Hi guys,

I notices that the OME-TIFF created by AICSImageIO use the OME-XML schema that is outdated. Can you update this to the latest or allow to choose the version?

toloudis, 2020-05-06:
We have been discussing a little reworking of the Writer for a 4.0 release very soon and I think that's a good idea.  I'm in favor of always writing the latest known schema if we have good evidence that the most popular readers can read it (ImageJ, Napari, ChimeraX, etc).  

Are you running into incompatibilities because of the schema version?

JacksonMaxfield, 2020-05-06:
Hey @sebi06, wanted to chime in as well. We somewhat know about the OME issue. Unfortunately there isn't the greatest resources in pure Python currently (to my knowledge) the [`omexml` object](https://github.com/CellProfiler/python-bioformats/blob/master/bioformats/omexml.py) hasn't been updated since 2018 which is the base of what we work with.

We are actively working (a PR right now :tada: ) on having a direct CZI to valid OME metadata converter tool but for the general statement of "build the OME XML from these parameters" is a bit of a ways out. As @toloudis said, the 4.0 writers release will be the likely candidate.

sebi06, 2020-05-07:
The thing is we have trouble to open the OME-TIFF created by your library in ZEN because we rely on strictyl on the OME scheme. Also also noticed that 

https://www.openmicroscopy.org/Schemas/OME/2013-06/      -  this is what should be used (i think)
https://www.openmicroscopy.org/Schemas/ome/2013-06/       - this is  what is used (fallback)?

In our case we run into issues when "ome" is used instead of "OME".

And a converter is is cool but does not solve our problem which is (and I also would really like not to convert CZItoOMETIFF beforehand (ZEN can do this anyway)):

- I start ZEN and load a CZI
- in run an Docker Container in ZEN to do some fancy processing using Python (or ...)
- the result inside the container is a NumPy stack
- since we do not have libCZIrw (published yet) and there is not Python warpper for that, we write an OME-TIFF based on the Numpy Array using your library
- the resulting OME-TIFF is "passed" from the docker container back to ZEN
- **ZEN tries to read those OME-TIFFs but cannot open them right now ...**
- the workflow should go on

This tool (APEER (on-site) has an UI but also the scripting interface.

[image]

Sidenote: If there would be libCZIrw, maybe even with a Python interface, would you consider storing your results as CZIs? Given the well-known limitations of OME-TIFF this sounds like a good option.

sebi06, 2020-05-07:
To further illustrate the issue. This OME-TIFF written by your library does not work for use

```xml
<?xml version='1.0' encoding='utf-8'?>
<OME xmlns="http://www.openmicroscopy.org/Schemas/ome/2013-06" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" UUID="urn:uuid:108d9fdd-d790-48e8-8877-ad3ff4a64756" xsi:schemaLocation="http://www.openmicroscopy.org/Schemas/OME/2013-06 http://www.openmicroscopy.org/Schemas/OME/2012-03/ome.xsd">
  <Image ID="0" Name="A01_S0000.ome.tiff">
    <AcquisitionDate>2020-05-07T15:44:10.194761</AcquisitionDate>
    <Pixels BigEndian="true" DimensionOrder="XYZCT" ID="0" PhysicalSizeX="0.457" PhysicalSizeY="0.457" PhysicalSizeZ="0.457" SizeC="1" SizeT="1" SizeX="1960" SizeY="1416" SizeZ="1" Type="int16">
<Channel ID="Channel:0:0" Name="AF568" SamplesPerPixel="1">
        <LightPath />
      </Channel>
    <TiffData FirstC="0" FirstT="0" FirstZ="0" IFD="0" PlaneCount="1" /></Pixels>
  </Image>
</OME>
```

But when I change compression to 0 in your code and then "manually" change the OME-TIFF it works fine in ZEN. So maybe a quick "dirty hack" would be to allowing setting the compression and changing the used XML schemas to the one below? I did not figure out how to do this in your code ...

```xml
<?xml version='1.0' encoding='utf-8'?>
<OME xmlns="http://www.openmicroscopy.org/Schemas/OME/2016-06" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" UUID="urn:uuid:108d9fdd-d790-48e8-8877-ad3ff4a64756" xsi:schemaLocation="http://www.openmicroscopy.org/Schemas/OME/2016-06 http://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd">
  <Image ID="0" Name="A01_S0000.ome.tiff">
    <AcquisitionDate>2020-05-07T15:44:10.194761</AcquisitionDate>
    <Pixels BigEndian="true" DimensionOrder="XYZCT" ID="0" PhysicalSizeX="0.457" PhysicalSizeY="0.457" PhysicalSizeZ="0.457" SizeC="1" SizeT="1" SizeX="1960" SizeY="1416" SizeZ="1" Type="int16">
<Channel ID="Channel:0:0" Name="AF568" SamplesPerPixel="1">
        <LightPath />
      </Channel>
    <TiffData FirstC="0" FirstT="0" FirstZ="0" IFD="0" PlaneCount="1" /></Pixels>
  </Image>
</OME>
```

toloudis, 2020-05-07:
Thank you for the extra details! That really helps us understand the issue.  We will be sure this is fixed.

toloudis, 2020-05-07:
Our library would certainly support adding a CZIWriter but it is not a use case that has come up until now.  Our strongest interest has been to write to the most interchangeable/open file formats.

JacksonMaxfield, 2020-05-07:
@sebi06 Does the compression have to be 0? Can you try just updating the metadata and loading it in ZEN?

sebi06, 2020-05-11:
Hmmm.... I changed compression to 2 (as a test) and modified your python package and did write an OME-TIFF.
But I still have to modify the header because it still contains ome "ome" instead of "OME" where I do not understand where this comes from. I think i corrected every entry from `Schemas/ome/2013-03` to `Schemas/OME/2016-06

But still one line in the TIFF headerhas the "ome"`. But compression = 2 seemes to work.

JacksonMaxfield, 2020-05-11:
@sebi06 I think it is safe for us to update the `ome` to `OME`. The compression stuff seems odd... FIJI opens all of our images just fine and we somewhat use that as our baseline "does it load".

sebi06, 2020-05-12:
I think oyu can leave the compression issue out of the discussion, because I do not think anymore that this is the cause of our issues. But updating the namespace to 2016-06 etc. would be cool.

JacksonMaxfield, 2020-05-12:
Cool. We have a patch already in place. Should be released hopefully within the next week.

heeler, 2020-05-13:
I just wanted to say thanks for all the discussion here, super useful and I'm fighting to get this into the transformation. We are hoping this will become something where the community at large can contribute and enable the conversion of their favorite metadata pieces.

closed 2020-05-13 by JacksonMaxfield

JacksonMaxfield, 2020-05-13:
Closed and patched in `aicsimageio==3.2.0`

commit 925583f1 (2020-05-13): Resolves #115, update OME spec
PR #116 (2020-05-13): bugfix/update-ome-spec
## Description
ZEN viewer fails to load our OME-TIFFs due to incorrect xmlns pointers and out of date schema. Following discussion on #115, the minimal changes of simply updating the xmlns pointers and schema version simply works. Tested in both ZEN and FIJI in addition to our normal OmeTiffWriter tests. This is very much a temporary patch until 4.0 writers release.

## Pull request recommendations:
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [g [... cut]