AllenCellModeling/aicsimageio #117: Reading Tiles (M-Index) from CZI images

sebi06, 2020-05-19:
Hi guys,

when I read a CZI I see the following log output

`Data has dimension M with depth 25, assuming M=0 is the desired value, if not the case specify M=x where x is an integer, list, tuple, range, or slice`

So does this imply that I just have to say ...., M=[0, 1, 2, 3] in order to read the first 4 tiles?

BTW the desired ideas is to either read the all tiles at once. Or Do I have to use aicspylibczi directly in this case and figure out the overlap topics with tiles by myself?

JacksonMaxfield, 2020-05-19:
Hey @sebi06,

So I think this should probably more prevalent in the README documentation, the `AICSImage` object is basically a "give me the most standard imaging data" (which for us is `STCZYX`). However, the underlying image readers do have access to their own implementations.

I would give `readers.CziReader` a try as it should have all the data and dimensions you want just not in "standard" order. (note it doesn't have the `get_image_data` function). If something seems like it is missing on the underlying reader class please let us know and I think we will be happy to figure out how to add stuff to them.

If that doesn't work, then yea I would say go to `aicspylibczi`.

heeler, 2020-05-19:
Hi @sebi06,

So from aicsimageio, I believe you should be able to use M as if it were a dimension but it will give you a data matrix with M as an index. 

What I'm guessing you want to do is use the libCZI mosaic reader. This is possible from aicspylibczi simply call the function read_mosaic from there https://allencellmodeling.github.io/aicspylibczi/aicspylibczi.html#aicspylibczi.CziFile.CziFile.read_mosaic.

One comment on the libCZI mosaic reader is that if the scale_factor is something other than 1 it struggles with the image, ie it drops a tile. I'm fairly sure this is happening from within libCZI but I haven't had time to debug it myself yet. A number of people have reached out to me asking me to troubleshoot this but I haven't had a chance yet. 

Hope this clarifies things. If other things pop up please let me know. I'm busy on a few other projects at the moment but I will be coming back to this to give it some more love in the near future.

Thanks, 
      ~Jamie

JacksonMaxfield, 2020-05-28:
@sebi06 Can I close this or do you think we need to add functionality?

sebi06, 2020-05-28:
Yes I consider this closed. I use pylibCZI directly for now.

closed 2020-05-28 by sebi06