AllenCellModeling/aicsimageio #127: is_pyramid deprecated in napari_0.3.0

mniederhuber, 2020-06-16:
## System and Software
* aicsimageio Version: 3.2.2
* Python Version: 3.7.7
* Operating System: macOS 10.14.6

## Description
`img = AICSImage("my.lif")`
`img.napari_view()`

results in:
`TypeError: view_image() got an unexpected keyword argument 'is_pyramid'
`

I think the `is_pyramid` parameter  of `napari.view_image()` has been removed per the [napari_0.3.0 release](https://github.com/napari/napari/blob/5c23220938488c01b876183b4a488acca55a2c36/docs/release/release_0_3_0.md)

closed 2020-06-16 by mniederhuber

JacksonMaxfield, 2020-06-16:
Hey @mniederhuber sorry didn't get back to you earlier. Thanks for alerting us. I will _try_ to put through a patch to fix this but I will also say that we are deprecating the `view_napari` function in the `4.0.0` release of `aicsimageio`. While not the exact same, you may be interested in [`napari-aicsimageio`](https://github.com/AllenCellModeling/napari-aicsimageio) which tells the "application" version of `napari` to use `aicsimageio` for all file reading.

mniederhuber, 2020-06-16:
Hi @JacksonMaxfield, thanks for the response. I did eventually come across your note about removing the `view_napari` function in the next release, hence why I closed the issue. The `napari-aicsimageio` plugin does the trick!