AllenCellModeling/aicsimageio #138: Slow reading of CZI files

KaSaBe, 2020-07-08:
## System and Software
* aicsimageio Version: 3.2.3
* Python Version: 3.7.6
* Operating System: Windows 10

## Description
Thank you for providing this package, which I am especially interested in using to create lazy dask arrays of a number of CZI files. Unfortunately, I am getting very slow performance using both imread and imread_dask from aicsimageio. The type of microscope or where the file is located (network drive or local drive) does not matter except for the IO limited reading of czifile, but for aicsimagio the time scales with file size. I tried with and without a dask_distributed.Client running, this did not make a difference either. The example file here is a ~600 MB CZI (3x60x1280x1280, CZYX) file.  Am I using it wrong?

## Reproduction
Actual file loading time using czifile (IO limited):
```
import czifile
%time img = czifile.imread(network_file_path)
Wall time: 6.43 s
%time img = czifile.imread(local_file_path)
Wall time: 306 ms
```
File loading time using aicsimageio:

```
import aicsimageio
%time img = aicsimageio.imread(network_file_path)
Wall time: 34.8 s
%time img = aicsimageio.imread(local_file_path)
Wall time: 32.2 s
```

Even constructing a dask array is much slower than simply loading the file with czifile:

```
%time img = aicsimageio.imread_dask(network_file_path)
Wall time: 15.4 s
%time img = aicsimageio.imread_dask(local_file_path)
Wall time: 13.5 s
```

Thank you for your help.

JacksonMaxfield, 2020-07-08:
Hey @KaSaBe!

Sorry to hear this is happening when we specifically run [benchmarks against `czifile` / `tifffile`](https://allencellmodeling.github.io/aicsimageio/benchmarks.html). From my end I am assuming we missed something in our benchmarks but not entirely sure.

Is there any chance I could get a copy of the file you are trying to load for my own testing? If you can't share I will try to find / ask a coworker for a ~600MB CZI with similar shape to yours later this week and do testing on it.

JacksonMaxfield, 2020-07-08:
Hey just wanted to update you. I am able to reproduce this behavior (unfortunately :( ).

We are going to track down if the issue is caused by our own CZI reader ([aicspylibczi](https://github.com/allencellmodeling/aicspylibczi)) or by `aicsimageio`. Current thinking is it is something to down with our own CZI reader which means there is hope to at the very least basically make the `imread` function route to `czifile` if necessary.

Either way, we are working on it!

JacksonMaxfield, 2020-07-08:
Confirmed to be `aicspylibczi`:

```
In [17]: %time czifile.imread("/allen/aics/microscopy/PRODUCTION/PIPELINE_4_OptimizationAutomation/3500002032/ZSD1/100X_PSF/3500002032_100X_20180511_psf-Scene-1-P1-G09.czi").shape           
CPU times: user 200 ms, sys: 376 ms, total: 577 ms
Wall time: 277 ms
Out[17]: (1, 1, 8, 30, 624, 924, 1)

In [18]: %time aicspylibczi.CziFile("/allen/aics/microscopy/PRODUCTION/PIPELINE_4_OptimizationAutomation/3500002032/ZSD1/100X_PSF/3500002032_100X_20180511_psf-Scene-1-P1-G09.czi").read_image
    ...: ()[0].shape                                                                                                                                                                          
CPU times: user 5.75 s, sys: 444 ms, total: 6.2 s
Wall time: 6.18 s
Out[18]: (1, 1, 8, 30, 624, 924)

In [19]: %time aicsimageio.imread("/allen/aics/microscopy/PRODUCTION/PIPELINE_4_OptimizationAutomation/3500002032/ZSD1/100X_PSF/3500002032_100X_20180511_psf-Scene-1-P1-G09.czi").shape       
/home/jacksonb/miniconda3/envs/test/lib/python3.7/site-packages/aicsimageio/transforms.py:178: UserWarning: Data has dimension B with depth 1, assuming B=0 is the desired value, if not the case specify B=x where x is an integer, list, tuple, range, or slice.
  f"Data has dimension {dim} with depth {data.shape[dim_index]}, "
CPU times: user 5.99 s, sys: 360 ms, total: 6.35 s
Wall time: 6.36 s
Out[19]: (1, 1, 8, 30, 624, 924)
```

Shortened:
```
czifile imread: 277ms
aicspylibczi imread (basically a no-dask check): 6.18s
aicsimageio imread (no dask): 6.36s
```

JacksonMaxfield, 2020-07-08:
For sanity and logging of the issue, I also ran time checks against the same image data but as an `OmeTiff`:

```
In [9]: %time tifffile.imread("/allen/programs/allencell/data/proj0/add/873/2a4/2fb/4c8/f79/742/86b/1a1/001/00/3500002032_100X_20180511_psf-Scene-1-P1-G09.ome.tiff").shape                   
CPU times: user 4.9 s, sys: 267 ms, total: 5.16 s
Wall time: 5.16 s
Out[9]: (30, 8, 624, 924)

In [10]: %time aicsimageio.imread("/allen/programs/allencell/data/proj0/add/873/2a4/2fb/4c8/f79/742/86b/1a1/001/00/3500002032_100X_20180511_psf-Scene-1-P1-G09.ome.tiff").shape               
CPU times: user 5.41 s, sys: 307 ms, total: 5.72 s
Wall time: 5.74 s
Out[10]: (1, 1, 8, 30, 624, 924)
```

Shortened:
```
tifffile imread: 5.16s
aicsimageio imread: 5.74s
```

What I would consider comparable (assumed extra time is from dask array creation)

JacksonMaxfield, 2020-07-08:
I will also be adding more CZI files to the benchmark dataset. Entirely my bad for not catching it earlier.

KaSaBe, 2020-07-09:
Thanks for looking into this! If it is of interest I tinkered a bit with lazy loading from czifile by stacking dask.delayed calls on the object (these are single-channel 2D images):

`czifile.CziFile.subblock_directory.data_segment().data `

This is not perfect and there might be better ways to do this, as I get read errors every now and then, but if I limit the slicing a bit it works all right. So this could be an alternative.

JacksonMaxfield, 2020-07-09:
Yea I have been digging through the `czifile` code as well. Iirc, one of the primary reasons we (really, @heeler) built out `aicspylibczi` was for more in depth handling of sub-blocks and mosaics and so our concern is that we would lose some functionality somewhere.

From discussion, @heeler said he was going to run some comparisons between the two libraries as to where they differ. I don't know if we are quite ready to drop `aicspylibczi` in favor of `czifile` because I remember there were very specific cases where we needed the lower level library but we will see after he runs the checks.

JacksonMaxfield, 2020-08-29:
Hey @KaSaBe just wanted to give you a very long overdue but nice update!

@heeler is actively working on this over on our CZI reading library! https://github.com/AllenCellModeling/aicspylibczi/pull/44

I don't have all the details but from my understanding it seems that once we get a new version of `aicspylibczi` out with the above changes + more, we will have on average ~25% faster speeds than `czifile`. @heeler is that correct?

So we are working! Just taking a while. Thanks for the patience.

heeler, 2020-08-31:
Hi @KaSaBe, Thanks for posting this issue. I just finished overhauling the code to make it more performant with CZI files. I think it will be more in-line with expectations once it is merged and released. Again I really do appreciate your bringing this to our attention. Thanks!

JacksonMaxfield, 2020-11-24:
@KaSaBe this should now be resolved on `aicspylibczi==2.7.5` and `aicsimageio==3.3.2`

Benchmarks:
[image]

([benchmarking utils found here](https://github.com/AllenCellModeling/aicspylibczi/tree/master/scripts))

closed 2020-11-24 by JacksonMaxfield