AllenCellModeling/aicsimageio #140: Support Reading and Writing of Pyramidal Formats

JacksonMaxfield, 2020-07-17:
## Use Case
*Please provide a use case to help us understand your request in context*

Many OME-Tiffs and CZI files are pyramidal where LODs have different array shapes. It would be great to support the reading of these one shape (scene) at a time and the writing of these through iterable of arrays.

## Solution
*Please describe your ideal solution*

`tifffile` _very recently_ [added support for both reading and writing of pyramidal tiffs](https://github.com/cgohlke/tifffile/commit/6b8fffe518b34ea787c51be6fd211e430b6a8324#diff-bd13ed7b3d0e1c6bfac4a7d96e8ae43a), pull in these changes and wrap in our API and `Reader` and future `OmeTiffWriter`.

Note, these will likely not be supported until after current institute high priority projects are done and after `aicsimageio==4.0.0` release.

JacksonMaxfield, 2020-12-14:
Waiting for test file data, pyramids of CZI and TIFF

JacksonMaxfield, 2021-01-26:
Adopt the `levels` API for reading:

```python
img = AICSImage("my-pyramid.czi")
img.levels  # returns tuple of levels
img.current_level  # returns current level, default / starts at level 0 (full resolution)
img.set_level  # update current level
```

Undecided how to adopt for Writing.

JacksonMaxfield, 2021-01-27:
Handle known bug of multi-resolution (multi-level) OME TIFFs selecting the wrong metadata due to "Images" being created _correctly_ for the total number of scenes in the file, but they don't use the matching IDs for the correct scenes / resolutions.

JacksonMaxfield, 2021-01-27:
Tracking `ome-types` metadata bug over on `ome-types`: https://github.com/tlambert03/ome-types/issues/71

JacksonMaxfield, 2021-01-27:
Going to work with Zarr devs to see if the chunked read issue on multi-resolution / pyramid is a:

* dask `from_zarr` implementation issue
* tifffile `ZarrTiffStore` spec implementation issue
* or a Zarr "the spec is still undetermined" issue

Reasoning here is that dask `from_zarr` is assuming there will be a key of just `./zarray` on the `MutableMapping` object and tifffile `aszarr` supports multi-resolution zarr by creating keys with names: `0/.zarray`, `1/.zarray`, etc. for the different resolution levels. This causes chunk read time issues when they key is requested and a `ValueError` occurs.

cgohlke, 2021-01-27:
```Python
In [4]: da.from_zarr(t.series[0].levels[0].aszarr())
<snip>
ValueError: not enough values to unpack (expected 2, got 1)
```
To access the first level of the first multi-resolution series as zarr array, not group, use `t.series[0].aszarr(level=0)`. This is currently necessary  because `t.series[0].levels[0]` is the same as `t.series[0]`.

JacksonMaxfield, 2021-01-27:
Thanks @cgohlke. Patched in the current open PR to default reading level 0 of pyramid files for now.

rcasero, 2021-07-06:
@JacksonMaxfield Reading of pyramid levels is not available in aicsimageio 4.0.3, is it?

The CZI files I provided for #274 can be used as pyramid examples, if that's useful

https://drive.google.com/drive/folders/1XukPEpCfIvibTT_6992sCgXeszxg-hUK?usp=sharing

Is there a way to open a CZI file with aicsimageio and save it to a TIFF or some other format with the pyramid?

rcasero, 2021-07-06:
> Is there a way to open a CZI file with aicsimageio and save it to a TIFF or some other format with the pyramid?

Replying to myself, I think maybe with `pyvips.Image.tiffsave()`, where it's possible to select an output with a pyramid. But here, it seems that what we'd do is load the largest level with `aicsimageio`, and then re-create a pyramid with `pyvips`, rather than copying over the already exisitng pyramid.

JacksonMaxfield, 2021-07-06:
@rcasero 4.0.3 does not support multi-scale / pyramid level selection. This issue is slated for 4.1.

You can use bioformats or pyvips but I am not an expert in either of them to give you the best guidance there.

rcasero, 2021-07-08:
@JacksonMaxfield Thanks. I made it work with `pyvips.tiffsave`

```
im_vips.tiffsave(tif_filename, tile=True, pyramid=True, resunit='cm',
                             xres=float(1e-3/xres), yres=float(1e-3/yres), bigtiff=True)
```

https://github.com/rcasero/cytometer/blob/a3f9abf792f24dba771d9940fb49c9bde25efc26/cytometer/data.py#L1618

closed 2023-04-30 by github-actions[bot]