AllenCellModeling/aicsimageio #141: KeyError: 'B' when trying to read shape of a CZI file

sebi06, 2020-07-21:
## System and Software
* aicsimageio Version: 3.2.4
* Python Version: 3.7.6
* Operating System: Windows 10

## Description

I get a KeyError when trying to read a CZI file:

```bash
(stardist) C:\Users\m1srh\Documents\GitHub\czi_demos>C:/Progages\aicsimageio\readers\czi_reader.py", line 380, in _daread_saferamData/Anaconda3/envs/stardist/python.exe c:/Users/m1srh/Do=chunk_by_dims, S=S)cuments/GitHub/czi_demos/test_aicsimageio.py                ages\aicsimageio\readers\czi_reader.py", line 252, in _daread
TYX0
(3, 1200, 1600, 3)
STCZYX
Traceback (most recent call last):
  File "c:/Users/m1srh/Documents/GitHub/czi_demos/test_aicsimageio.py", line 15, in <module>
    print(czi_aics.shape)
  File "C:\ProgramData\Anaconda3\envs\stardist\lib\site-packages\aicsimageio\aics_image.py", line 238, in shape
    return self.size()
  File "C:\ProgramData\Anaconda3\envs\stardist\lib\site-packages\aicsimageio\aics_image.py", line 228, in size
    return tuple([self.dask_data.shape[self.dims.index(dim)] for dim in dims])
  File "C:\ProgramData\Anaconda3\envs\stardist\lib\site-packages\aicsimageio\aics_image.py", line 228, in <listcomp>
    return tuple([self.dask_data.shape[self.dims.index(dim)] for dim in dims])
  File "C:\ProgramData\Anaconda3\envs\stardist\lib\site-packages\aicsimageio\aics_image.py", line 178, in dask_data
    reader_data = self.reader.dask_data
  File "C:\ProgramData\Anaconda3\envs\stardist\lib\site-packages\aicsimageio\readers\reader.py", line 156, in dask_data
    self._dask_data = self._read_delayed()
  File "C:\ProgramData\Anaconda3\envs\stardist\lib\site-packages\aicsimageio\readers\czi_reader.py", line 397, in _read_delayed
    self._file, chunk_by_dims=self.chunk_by_dims, S=self.specific_s_index
  File "C:\ProgramData\Anaconda3\envs\stardist\lib\site-packages\aicsimageio\readers\czi_reader.py", line 387, in _daread_safe
    raise e
  File "C:\ProgramData\Anaconda3\envs\stardist\lib\site-packages\aicsimageio\readers\czi_reader.py", line 380, in _daread_safe
    return CziReader._daread(img=img, czi=czi, chunk_by_dims=chunk_by_dims, S=S)
  File "C:\ProgramData\Anaconda3\envs\stardist\lib\site-packages\aicsimageio\readers\czi_reader.py", line 252, in _daread
    image_dim_indices[dim][1] - image_dim_indices[dim][0]
KeyError: 'B'
```

## Reproduction

Run code snippet on the provided CZI file (I had to attach it as a ZIP ...)
[Tumor_H&E_uncompressed_TSeries_3.zip](https://github.com/AllenCellModeling/aicsimageio/files/4951843/Tumor_H.E_uncompressed_TSeries_3.zip)

```python
from aicsimageio import AICSImage, imread, imread_dask
import czifile as zis

filename = r"C:\temp\
[Tumor_H&E_uncompressed_TSeries_3.zip](https://github.com/AllenCellModeling/aicsimageio/files/4951848/Tumor_H.E_uncompressed_TSeries_3.zip)
Tumor_H&E_uncompressed_TSeries_3.czi"

# get CZI object
czi = zis.CziFile(filename)
# add axes and shape information using czifile package
print(czi.axes)
print(czi.shape)

# add axes and shape information using aicsimageio package
czi_aics = AICSImage(filename)
print(czi_aics.dims)
print(czi_aics.shape)
print(czi_aics.size_x)
print(czi_aics.size_y)
print(czi_aics.size_c)
print(czi_aics.size_t)
print(czi_aics.size_t)
print(czi_aics.size_s)

czi.close()
czi_aics.close()
```

JacksonMaxfield, 2020-07-21:
Hey @sebi06 thanks for reporting the issue. Will try to resolve it later today as I think I know what is likely the issue. Will keep you posted if the issue seems like more than just a day long bugfix.

JacksonMaxfield, 2020-07-21:
Hey @sebi06 what do you expect the dimensions of this image to be out of curiosity? `czifile` returns dims `TYX0` and `aicspylibczi` returns dims `BTCYX`

JacksonMaxfield, 2020-07-21:
@sebi06 I am handing this over to @heeler. After a lot of investigating this seems very much like an issue with how we read BGR images in `aicspylibczi`. If we tried to fix it on `aicsimageio`'s side, the change would have to be placed in release 4.0 as it is a breaking change on this side.

@heeler, the main issue at hand is that currently `aicspylibczi.CziFile.dims_shape` doesn't return any `C` or `B` information while after reading a chunk of the image as a sample, all dimensions are "found" and thus our dask array construction fails (transposing and getitems don't know which array indicies to access because mismatched or entirely different dims are present)

JacksonMaxfield, 2020-08-29:
Hey @sebi06, to my knowledge this will be patched or a better solution will be available on the next `aicspylibczi` release! @heeler is currently optimizing reading times as well as BGR support here: https://github.com/AllenCellModeling/aicspylibczi/pull/44

So we are working! Just got a lot going on. Thanks for the patience!

JacksonMaxfield, 2020-11-11:
@sebi06 do you know if I can mark this resolved after using the newer `aicspylibczi` versions?

sebi06, 2020-11-12:
We I try to re-run the code snippet I provided above I still get an error, but now it is a different one:

```text
(imageanalysis) C:\Users\m1srh\Documents\GitHub\czi_demos>C:/ProgramData/Anaconda3/envs/imageanalysis/python.exe c:/Users/m1srh/Documents/GitHub/czi_demos/test_keyerror_B.py
TYX0
(3, 1200, 1600, 3)
STCZYX
Traceback (most recent call last):
  File "c:/Users/m1srh/Documents/GitHub/czi_demos/test_keyerror_B.py", line 17, in <module>
    print(czi_aics.shape)
  File "C:\ProgramData\Anaconda3\envs\imageanalysis\lib\site-packages\aicsimageio\aics_image.py", line 239, in shape
    return self.size()
  File "C:\ProgramData\Anaconda3\envs\imageanalysis\lib\site-packages\aicsimageio\aics_image.py", line 229, in size
    return tuple([self.dask_data.shape[self.dims.index(dim)] for dim in dims])
  File "C:\ProgramData\Anaconda3\envs\imageanalysis\lib\site-packages\aicsimageio\aics_image.py", line 229, in <listcomp>
    return tuple([self.dask_data.shape[self.dims.index(dim)] for dim in dims])
  File "C:\ProgramData\Anaconda3\envs\imageanalysis\lib\site-packages\aicsimageio\aics_image.py", line 179, in dask_data
    reader_data = self.reader.dask_data
  File "C:\ProgramData\Anaconda3\envs\imageanalysis\lib\site-packages\aicsimageio\readers\reader.py", line 156, in dask_data
    self._dask_data = self._read_delayed()
  File "C:\ProgramData\Anaconda3\envs\imageanalysis\lib\site-packages\aicsimageio\readers\czi_reader.py", line 397, in _read_delayed
    self._file, chunk_by_dims=self.chunk_by_dims, S=self.specific_s_index
  File "C:\ProgramData\Anaconda3\envs\imageanalysis\lib\site-packages\aicsimageio\readers\czi_reader.py", line 387, in _daread_safe
    raise e
  File "C:\ProgramData\Anaconda3\envs\imageanalysis\lib\site-packages\aicsimageio\readers\czi_reader.py", line 380, in _daread_safe
    return CziReader._daread(img=img, czi=czi, chunk_by_dims=chunk_by_dims, S=S)
  File "C:\ProgramData\Anaconda3\envs\imageanalysis\lib\site-packages\aicsimageio\readers\czi_reader.py", line 252, in _daread
    image_dim_indices[dim][1] - image_dim_indices[dim][0]
KeyError: 'C'
````

JacksonMaxfield, 2020-11-12:
Does this happen to be a multi-channel BGR CZI?

sebi06, 2020-11-12:
Yes.

JacksonMaxfield, 2020-11-12:
Ahhhhh, that explains it. We don't currently handle multi-channel RGB / BGR images in aicsimageio 3.*, but in 4.0 they _should_ be supported. I just got multi-channel [RGB TIFFs working yesterday](https://github.com/AllenCellModeling/aicsimageio/pull/160).

@heeler did we ever decide if multi-channel BGR CZIs were going to be supported by `aicspylibczi`? I think now that we can handle them in `aicsimageio>=4.0.0` we should support. Would simply mean returning the "samples" dimension with the `dims_shape` iirc.

sebi06, 2020-11-12:
OK, I was not aware of it. I thing BGR is supported in aicspylibczi, if I am not mistaken.

heeler, 2020-11-12:
Multichannel BGRs are handled. Essentially you get a 3x the number of channels back for a multichannel BGR. 
I'm not 100% but if I remember right if you specify C=2 it would be C=2 of the BGR so you would get 3 channels back. 
We will probably have to polish this up a bit to get ideal behavior but the functionality is present.

JacksonMaxfield, 2020-11-12:
Thanks for the info @heeler 

@sebi06 I will update you probably in two weeks when I get to the 4.0 implementation of `CziReader`.

closed 2021-05-31 by JacksonMaxfield

JacksonMaxfield, 2021-05-31:
Hey @sebi06 this has now made it into the dev releases for 4.0! :tada: 

`pip install aicsimageio --pre` will get you the latest dev release.

```python
from aicsimageio import AICSImage

img = AICSImage("multi-channel-bgr.czi")
img.dims  # T C Z Y X S

# C is channel
# S is samples for BGR data
```

If you have the time and a desire to give it a go please do and please let us know how it goes.

Please note that the 4.0 API is a bit different from the 3.0 API so best to check the README / docs for changes there too.

If you have any issues please just post them as a new GitHub issue.

sebi06, 2021-05-31:
This looks good. I will have a look soon.
One questions that came to my mind is about that S = Samples.

But what happened to S = Scenes? 

Are scenes completely removed from the dimension string in aicsimageio now?

toloudis, 2021-05-31:
The AICSImage class has separated "scene" out from the standard 5D OME dimensions.  We now have AICSImage.scenes and AICSImage.set_scene(scene_name)

PR #231 (2021-05-31): feature/czi-reader
## Description
This implements the 4.0 support for Zeiss Czi Files in the class CziReader.
For 4.0 local file reading will only be supported, I hope to add remote reading for 4.1. 
lxml has been replaced with XML for serialization. 
Mosaic file stitching is in. 

There are tests for reading M, S(or A at the lower level) and for retrieving only part of the required chunk dimensions (X, Y, S)=( : , : , 0 ).

Resolves #218 
Resolves #163 
Resolves #141 
Resolves #62 
Resolves #239