AllenCellModeling/aicsimageio #159: Method get_image_data is mixing up dimensions when reading a CZI file

sebi06, 2020-11-05:
## System and Software
* aicsimageio Version: 3.3.1
* Python Version: 3.7.4
* Operating System: Windows 10

## Description

I have a CZI image (attached as a zip) which I want to process. It is basically a timeseries (T=3), where every timepoint is  a Z-Stack (Z=5) with 2 channels (C=2). I need to access zstack per timepoint per channel and come up with. I did include some lines to visualize my issue (show napari) which are just there to illustrate my problem.

```python
img = AICSImage(image_path)

for t in range(img.size_t):
    for ch in range(img.size_c):

        # read a zstack
        zstack = img.get_image_data("ZYX", S=0, T=t, C=ch)
        print('Showing TimePoint - Channel : ', t, ch)
        print('Shape Z-Stack : ', zstack.shape)

        with napari.gui_qt():
            viewer = napari.view_image(zstack, rgb=False)

        # loop over single z-planes
        for z in range(img.size_z):

            # do processing ...
        ...
```
Here is what I see when I check the stack (using napari) and compare it to reading the full image (using napari). For the 1st z-plane it looks OK

[image]

And now navigate to Z=3 inside the channel showing the chromosomes and suddenly I get a plane that contains image data from the wrong channel.

[image]

Since the stack looks OK when loading it completely inside Napari I suspect that something is wrong with:

`zstack = img.get_image_data("ZYX", S=0, T=t, C=ch)`

my expectation is that this would extract a zstack for S=0, T== and C=0 and also the output inside the console looks ok.

```text
Showing TimePoint - Channel :  0 0
Shape Z-Stack :  (5, 170, 240)
```
So it did extract a stack with 5 planes, but just plane Z=3 has the wrong content. If one navigates to Z=4 the content looks OK again. Any idea what my mistake is or is this a bug?

[CellDivision_T=3_Z=5_CH=2_X=240_X=170.zip](https://github.com/AllenCellModeling/aicsimageio/files/5492616/CellDivision_T.3_Z.5_CH.2_X.240_X.170.zip)

sebi06, 2020-11-05:
Hi all, I digged a litte depper and tried to read the zstack by using aicspylibczi directly. And here the data are correct. No mixing of channels.

So does this mean the problem after reading the CZI via `aicspylibczi` somewhere inside the `aicsimageio` library?

```python
from aicspylibczi import CziFile

czi = CziFile(image_path)
img, shp = czi.read_image(S=0, T=0, C=0)
with napari.gui_qt():
    viewer = napari.view_image(img, rgb=False)
```

[image]

heeler, 2020-11-05:
Hi @sebi06 I'll look into this today and see if I can figure out what's going on. Thanks for putting in the issue!

heeler, 2020-11-05:
@sebi06 What version # of aicspylibczi where you running with this? Could you put the contents of `pip freeze` in here so we know the working environment in case that is a factor. Thanks!

sebi06, 2020-11-05:
I think it was the latest aicspylibczi=2.7.1.

Using this one it works fine.

heeler, 2020-11-05:
I'm trying to confirm that I haven't introduced an error in the 2.7.1 release. To me when I go through it carefully in fiji it doesn't look right if I use 2.7.1 but if I uninstall aicspylibczi and do a `pip install aicspylibczi==2.6.0` and restart jupyter etc things look right. If you could confirm this that would be helpful as I don't know the ground truth of the image.

sebi06, 2020-11-06:
Hi @heeler ,

maybe I misunderstand you. Here is what I did

- tried to read the CZI using AICSImageIO --> **this causes mixed planes**
- now I use aicspylibczi directly to read the CZI --> this works fine for version 2.7.1
- I downgraded to version 2.6.0
- i used aicspylibczi to read the CZI and this works fine

So doesn't this mean, that the problem is inside AICSImageIO rather than inside aicspylibczi, because there I never had problems so far. Below a screenshot from using aicspylibczi directly.

[image]

heeler, 2020-11-06:
Hi @sebi06, I have a patch that I'm working on that I believe fixes it. It's being reviewed right now and I hope to have it published to pypi some time today. Hopefully this will resolve this issue. I'll put a comment in here when it's up and likely ask you to confirm the it works for you.

heeler, 2020-11-06:
Hi @sebi06, I know your test didn't seem to indicate aicspylibczi but in my hands it did. I have a patch I just published that I believe fixes it. In your python environment of choice `pip install --upgrade aicspylibczi` and then if using jupyter lab restart jupyter and give it a try. If that doesn't fix this for you please let me know. Thanks!

sebi06, 2020-11-06:
OK. But I could only see the issue when using aicsimageio.
Neither 2.6.0 or 2.7.1 of aicspylibczi gave any wrong results.

Let's see.

JacksonMaxfield, 2020-11-11:
Any issues since upgrading or can I mark as resolved?

sebi06, 2020-11-12:
I use now

* aicsimageio: 3.3.1
* aicspylibczi: 2.7.3

And here I have no issues when reading the CZI using AICSImageIO. Therefore I think the fix works. I will close the issue.

Thank so much.

closed 2020-11-12 by sebi06