AllenCellModeling/aicsimageio #161: Cores exception requested  but only  available.

snowformatics, 2020-11-12:
Hi,

I am trying to use aicsimageio to read a CSV file but I get only this message and the program stop to run:

`img = AICSImage( 'test.czi')
print (img.shape)`

Error:
Cores exception requested 71 but only 36 available.

Thanks in advance

JacksonMaxfield, 2020-11-12:
Hello @snowformatics 

I need a bit more information to help debug, could you please tell us:

* aicsimageio Version:
* Python Version:
* Operating System:

* The full code snippet you are trying to run:

Laslty, a clarification, are you trying to read a `CSV` or a `CZI` file? You say CSV but in your small snippet you have `test.czi`.

JacksonMaxfield, 2020-11-12:
The full traceback would be especially useful too!

toloudis, 2020-11-12:
sounds like the estimation in aicspylibczi of the number of cores available is not working in this particular case.  So knowing something about the OS and CPU you are using might help.

snowformatics, 2020-11-12:
I am sorry about my bad report, here are some more informations:

Python 3.6
Windows 10 Enterprise 2016 LTSB
Intel Xeon CPU E5-2695 v4
aicsimageio  3.3.1

Of course I meant CZI, sorry I edited my comment. There is no traceback just a print out. I am usually using czifile library directly  but I would like to try out and compare aicsimageio.

Thanks

heeler, 2020-11-12:
Hi @snowformatics thanks for reporting this it seems to be an odd difference only on windows between python and C++ queries as to how many cores there are available. I just published a new patch to aicspylibczi that should address this. If you could try it out that would be great. To do so in your python environment upgrade aicspylibczi via `pip install --upgrade aicspylibczi`. It should install aicspylibczi 2.7.4 which contains the patch. If you wouldn't mind letting me know how it goes that would be great. Thanks!

snowformatics, 2020-11-13:
> Hi @snowformatics thanks for reporting this it seems to be an odd difference only on windows between python and C++ queries as to how many cores there are available. I just published a new patch to [...]

Thanks your patched worked! Everything looks good now.

JacksonMaxfield, 2020-11-13:
Glad to hear everything worked! Marking this as resolved.

closed 2020-11-13 by JacksonMaxfield