AllenCellModeling/aicsimageio #162: Documentation or example with skimage / openCV

snowformatics, 2020-11-14:
Hi,

would it be possible to extent the documentation with a simple example on how to use aicsimageio together with skimage or openCV? It's not clear to me how to work most efficient with the data (multiporocesing, dask, dask-image). It would be very helpful to have an example for a very simple image processing pipeline, e.g. read image, apply threshold, write image with parallel processing.

Thank you very much for your efforts with aicsimageio,

best regards

JacksonMaxfield, 2020-11-14:
I don't know if _this_ library is the best place to have that documentation. I see the value in it, but, `aicsimageio` is purely about IO.

My thought process here is that while we return `dask` and `numpy` data types, it isn't this library's job to explain those data types as personally I would probably get some of the finer details wrong and I don't want to provide incorrect information.

We do have a [Performance Considerations](https://github.com/AllenCellModeling/aicsimageio#performance-considerations) section that somewhat addresses this question.

Lastly, and this is related to your other issue (#163), we are in the process of massively upgrading the library, that will allow the reading of BGR and more file types in general. I personally can see us providing examples like the one you proposed for the `4.0` release but not our current release.

That said, here are two examples (using current `aicsimageio==3.3.1`) of making max projections of a Z stack, one with `numpy` and one with `dask`. They are two examples because there are choices you as the developer need to make when deciding which to use. I provided the duration for each but notably these both ran on a MacBook Pro with 8 cores.

<details>
  <summary>Numpy Example (Click to Expand)</summary>

Script Duration: 14.57 seconds

```python
from aicsimageio import AICSImage
from aicsimageio.writers import OmeTiffWriter
import numpy as np

# Create AICSImage object
img = AICSImage("example.ome.tiff")

# Select only GFP and H3342 channels
selected_channels = [
    img.get_channel_names().index("EGFP"),
    img.get_channel_names().index("H3342"),
]

# Process each channel
processed_channels = []
for c in selected_channels:
    data = img.get_image_data(
        "ZYX",
        C=c,
        S=0,
        T=0,
    )
    
    # Get percentile norm by values
    min_px_val, max_px_val = np.percentile(
        data.flatten(),
        [50.0, 99.8],
    )

    # Norm
    normed = (data - min_px_val) / (max_px_val - min_px_val)

    # Clip any values outside of 0 and 1
    clipped = np.clip(normed, 0, 1)

    # Scale them between 0 and 255
    scaled = clipped * 255

    # Create max project
    proj = scaled.max(axis=0)
    
    # Append to processed list
    processed_channels.append(proj)
    
# Stack to numpy
processed_channels = np.stack(processed_channels)

# Write out new image
with OmeTiffWriter("processed.ome.tiff") as writer:
    writer.save(
        processed_channels,
        channel_names=[
            "max_projected_structure",  # GFP
            "max_projected_dna",  # H3342
        ],
        dimension_order="CYX",
    )
   ```
</details>

<details>
  <summary>Dask Example (Click to Expand)</summary>

Script Duration: 12.52 seconds

```python
from aicsimageio import AICSImage
from aicsimageio.writers import OmeTiffWriter
import dask.array as da
from distributed import Client, LocalCluster

# Create Distributed Client and Cluster
cluster = LocalCluster()
print(f"Dashboard at: {cluster.dashboard_link}")
client = Client(cluster)

# Create AICSImage object
img = AICSImage("example.ome.tiff")

# Select only GFP and H3342 channels
selected_channels = [
    img.get_channel_names().index("EGFP"),
    img.get_channel_names().index("H3342"),
]

# Process each channel
processed_channels = []
for c in selected_channels:
    data = img.get_image_dask_data(
        "ZYX",
        C=c,
        S=0,
        T=0,
    ).persist()
    
    # Get percentile norm by values
    min_px_val, max_px_val = da.percentile(
        data.flatten(),
        [50.0, 99.8],
    ).compute()

    # Norm
    normed = (data - min_px_val) / (max_px_val - min_px_val)

    # Clip any values outside of 0 and 1
    clipped = da.clip(normed, 0, 1)

    # Scale them between 0 and 255
    scaled = clipped * 255

    # Create max project
    proj = scaled.max(axis=0)
    
    # Append to processed list
    processed_channels.append(proj)
    
# Stack to numpy
processed_channels = da.stack(processed_channels)

# Write out new image
with OmeTiffWriter("processed.ome.tiff") as writer:
    writer.save(
        processed_channels.compute(),
        channel_names=[
            "max_projected_structure",  # GFP
            "max_projected_dna",  # H3342
        ],
        dimension_order="CYX",
    )
```
</details>

In short, I would stick to `numpy` (i.e. `get_image_data` and `data`) for simplicity, but, if you know the `dask` API well enough _and_ your `dask` cluster is large enough, there are cases where it can run quicker.

Hope all of this helps.

snowformatics, 2020-11-16:
I fully understand and agree with your statement! And thanks for the very helpful code examples.

JacksonMaxfield, 2020-11-16:
Glad I could help!

closed 2020-11-16 by JacksonMaxfield