AllenCellModeling/aicsimageio #163: Question regarding reading a czi file

snowformatics, 2020-11-14:
Hi again,

I have a question regarding reading a CZI file. czifile.py library gives me the shape  `(2, 1, 10, 22800, 28960, 3)` of my czi file. When I use aicsimageio  I get this shape:

`img = AICSImage('my.czi')`

Shape:
`(1, 1, 3, 10, 1200, 1600)`

I have two questions:
1) How to read the entire image and not only a part of it?
2) How to extract the number of scenes in the slide? 'S' refers to 1 but I have two scenes in the slide.

Thank you very much in advance

heeler, 2020-11-14:
What can you tell us about the file? 

Is it a mosaic file? If so you might be better off using aicspylibczi to access it. 

It sounds like it might have pixeltype RGB as well but I'm guessing. I'm assuming you mean Gohlke's czifile.py. I believe he treats the RGB as Samples whereas I treat them as channels so the shape comes out somewhat different in the 2 implementations.

I don't know if you've found the [docs pages](https://allencellmodeling.github.io/aicsimageio/index.html) yet but they can be helpful and we do try to make them informative. The default order is “STCZYX” for aicsimageio. It doesn't currently explain the handling of RGB images as in this case that's actually being done by aicspylibczi. 

The [.shape](https://allencellmodeling.github.io/aicsimageio/aicsimageio.html#aicsimageio.aics_image.AICSImage.shape) might be of use as well as the individual properties [size_s](https://allencellmodeling.github.io/aicsimageio/aicsimageio.html#aicsimageio.aics_image.AICSImage.size_s) size_z size_c etc. 

To retrieve the data in one go using [.data](https://allencellmodeling.github.io/aicsimageio/aicsimageio.html#aicsimageio.aics_image.AICSImage.data), assuming the shape of the data is consistent across channels (otherwise you have to specify the scene you want and deal with the 2 Scenes separately. 

Hope this helps.

JacksonMaxfield, 2020-11-14:
Clarification on @heeler's comment:

> The AICSImage object will only pull the Scene, Time, Channel, Z, Y, X dimensions from the reader. If your file has dimensions outside of those, use the base reader classes `CziReader`, `OmeTiffReader`, `TiffReader`, or `DefaultReader`.

In this case, it may be worth giving `aicsimageio.readers.CziReader` a try.

However, I do agree with @heeler, just based off shape alone. It looks like you have a BGR CZI file. `aicsimageio` doesn't support BGR files yet unfortunately.

JacksonMaxfield, 2020-11-14:
And, further expanding. @heeler made a great comment about "if you scenes are the same shape". If you have different shaped / sized scenes, you will need to specify which scene you want to read when you load the object.

`img = AICSImage("my.czi", S-0)` or `img = AICSImage("my.czi", S=1)` for example.

This process will be much easier to manage come release 4.0.

snowformatics, 2020-11-16:
Thanks a lot for your comments. Indeed I am working with mosaic RGB files. I will try again as soon as release 4.0 is out, looking forward for it and thanks for offering this as open source library. A short comment, at least in my case it is not known before how many scenes are on a slide.

In case you need some beta testing on RGB files, just let me know.

heeler, 2020-11-16:
If you are working with mosaic czi files I would suggest working with aicspylibczi directly. This is the library that does CZI reading for AICSImageIO but by dealing with it directly you'll get support to reassemble your mosaic files automatically. There is a function on the CziFile class defined in aicspylibczi called [read_mosaic()](https://allencellmodeling.github.io/aicspylibczi/aicspylibczi.html#aicspylibczi.CziFile.CziFile.read_mosaic) that should give you the reassembled image. Hope this helps.

JacksonMaxfield, 2020-11-16:
So there is a lot to unpack in this single GitHub issue and it spawned a lot of great questions in our dev meeting.

1. You are reading a multi-scene file
2. You are reading a mosiac file
3. You are reading an RGB file

---

1. Multi-scene will be much better handled by `aicsimageio==4.0.0`
2. We discussed the difference between `czifile` and `aicsimageio` and how `czifile` "reconstructs" the full mosaic image. Two things are happening here to cause the difference.
  1. The `AICSImage` object _only_ pulls the `S`, `T`, `C`, `Z`, `Y`, and `X` dimensions from the base `CziReader`. So while the `CziReader` will have an `M` dimension for "Mosaic tile", the `AICSImage` object is only pulling the first dimension (`M=0`). But, this alone doesn't explain the behavior.
  2. `czifile` is "reconstructing" the mosaic image into one large YX RGB plane where as our `CziReader` has the `M` dimension to select individual tiles, which may not be ideal. To resolve this, we can't say when, but we did discuss that at some point it would make sense to do this operation for the user just like `czifile` is doing. As @heeler said, you can use `aicspylibczi.read_mosaic` if you want the same functionality.
3. RGB volumetric data is a doozy. Different standard amongst the formats. We know how we are going to better handle this and have made a decision for `4.0`.

---

These are mostly just notes for us for later but also for you if you are curious. And to that end, I am going to leave this issue open until we put more thought or implementation into "reconstructing" the full mosaic image directly as functionality of `aicsimageio`.

JacksonMaxfield, 2020-11-16:
I have also changed the name of your issue to better reflect the current known issues with reading it.

toloudis, 2020-11-16:
I'll just chime in here to say that we probably should change the aicsimageio czi reader to automatically reconstruct the full mosaic image, as default behavior.  I'm not sure what that implies for chunked reads etc, and we might be relying on libCZI to be smart enough to handle that internally.   

Once we know we are reading the full image, which we consider the main use case, we can then add some extra api (maybe already exists in aicspylibczi?) to get the individual mosaic tiles out for users who need that.

MosGeo, 2021-01-20:
I will add my two cents here. I am currently reading a CZI file that is multichannel and is a mosiac. `aicspylibczi` is able to read the file perfectly as of 2.8.0 thanks to @heeler but I am thinking of using aicsimageio to utilize dask (my images are GBs in size).

Using the CziReader directly as suggested here results in reading the tiles dimensions but it seems to only read one channel. I am still investigating the issue. I will also try to create a simple example with the same characteristics to add to the tests in `aicsimageio `and `aicspylibczi`.

JacksonMaxfield, 2021-01-20:
Hey @MosGeo

Unfortunately `aicsimageio` 3.x doesn't support reading RGB / BGR CZI images.
I would say your specific image of "BGR" + multichannel + mosaic is one we are actively trying to support in aicsimageio in the next release (4.x).

We aren't too far out for that to be honest, maybe a couple of weeks? But yea just know we don't support it _yet_. We are working on it!

JacksonMaxfield, 2021-01-20:
Related, in supporting those images, we are actually planning an `aicspylibczi` major release to its own `3.0.0` to make some changes to align with `aicsimageio`

MosGeo, 2021-01-21:
@JacksonMaxfield Thanks for the update. This is awesome news. I look forward to aicsimageio 4 and aicspylibczi 3.

p.s. I have been looking at the source code of the project and you guys did a great job at laying the foundation. It is very accessible and clean. After I learn more and depending on my time commitments at the time,, I might be able to contribute if there are some small issues open.

JacksonMaxfield, 2021-01-21:
Thanks for the kind words. I personally think we have gotten even better about managing the codebase in 4.x too so hopefully it will be even easier to help out if you do end up having the time! :slightly_smiling_face:

closed 2021-05-31 by JacksonMaxfield

JacksonMaxfield, 2021-05-31:
Hey @snowformatics @MosGeo it happened -- CZI mosaic file stitching is now in the 4.0 dev release! :tada: 

```python
from aicsimageio import AICSImage

img = AICSImage("tiled.czi")
img.data  # stitched numpy array
img.dask_data  # stitched dask array
# each chunk is a couple of tiles so can load portions of the image instead of the whole mosaic

img.reader.data  # unstitched numpy array (with 'M' dimension for tiles)
img.reader.dask_data  # unstitched dask array (with 'M' dimension for tiles)
```

If either of you have time and a desire to check it out, please do! The docs are live and the README has some details. The API has changed a decent bit from 3.* to 4.0 so please be aware of that.

To install: `pip install aicsimageio --pre`

If you have any issues please let us know by making a new GitHub issue!

snowformatics, 2021-06-01:
That's awsome,  thanks a lot for your efforts!  I will test ASAP and give feedback!

MosGeo, 2021-06-02:
@JacksonMaxfield Awesome! I will give it a go. Excited about V4. Also, already excited about V4.1 (specifically #140) as it will make it possible to make fast web image viewers with rest api. 

p.s. Thanks for the merge for the small fix. This was my first official contribution to open source public repository where I am not the owner :)

JacksonMaxfield, 2021-06-02:
> Awesome! I will give it a go. Excited about V4. Also, already excited about V4.1 (specifically #140) as it will make it possible to make fast web image viewers with rest api.

Zarr will be the best for that and things like `viv` already exist and are _excellent_. Highly recommend checking it out: https://github.com/hms-dbmi/viv

And awesome. I myself need to make more general open source contributions haha.

PR #231 (2021-05-31): feature/czi-reader
## Description
This implements the 4.0 support for Zeiss Czi Files in the class CziReader.
For 4.0 local file reading will only be supported, I hope to add remote reading for 4.1. 
lxml has been replaced with XML for serialization. 
Mosaic file stitching is in. 

There are tests for reading M, S(or A at the lower level) and for retrieving only part of the required chunk dimensions (X, Y, S)=( : , : , 0 ).

Resolves #218 
Resolves #163 
Resolves #141 
Resolves #62 
Resolves #239