AllenCellModeling/aicsimageio #164: Dev Meeting Agenda 2020/11/15

JacksonMaxfield, 2020-11-16:
* [ ] Open PRs
  * [ ] feature/add-tiff-reader #160 
    * [ ] Defaults for `PhysicalPixelSizes`
    * [ ] Resolving the `tifffile` and `aicspylibczi` spec for multi-channel RGB images
    * [ ] Other comments
  * [ ] admin/update-base-reader-dep-version  #156 
    * [ ] Needs `aicspylibczi` dep upgrade
* [ ] Open Issues
  * [ ] Examples w/ processing of images #162 
  * [ ] More CZI documentation / wait until 4.0 #163
* [ ] Options for HDF5 + XML / `BigStitchedReader` [Thread](https://twitter.com/DougPShepherd/status/1328056264580947973)
* [ ] `OmeTiffReader` up-next

toloudis, 2020-11-16:
note: RGB is preferred to be handled as a separate "S" dimension (S for "samplesPerPixel")

JacksonMaxfield, 2020-11-16:
Notes:

@heeler to finish the `admin/update-base-reader-dep-versions`
* Set current `aicsimageio` (`3.3.1`) `aicspylibczi` dep to `>=2.7.5,<3`

@heeler to change `aicspylibczi` to use a "Samples" or related name dimension for RGB data instead of current expansion / interleaving into `Channels` dimension (this will be a breaking `aicspylibczi` change resulting in `aicspylibczi==3.0.0`). Our statement moving forward is that RGB is preferred to be handled as separate "S" dimension.

@JacksonMaxfield to resolve leftover issues

@JacksonMaxfield to resolve `feature/add-tiff-reader` PR

---

Bigstitcher and other "Mosiac" formats:
At some point in the future (potentially `4.1`) we should have the default behavior of files with "mosiac" elements be to reconstruct the image as a part of the `data` process / attribute. If users don't want this functionality it can be toggled by a `AICSImage` or `Reader` parameter.

---

Roadmap to 4.0:
* Volumetric Readers
  * `OmeTiffReader`
  * `CziReader`
  * `LifReader`
  * `ZarrReader` (Potentially)
* Writers
  * `DefaultWriter` (Direct copy over from existing in `release-4.0` branch)
  * `OmeTiffWriter` (Changes need to adopt `ome-types` and `multi-scene` support)
  * `OmeZarrWriter` (Potentially)

closed 2020-11-16 by JacksonMaxfield

toloudis, 2020-11-16:
In addition to stating that RGB will be handled as a "S" dimension, I would say this:
If samples per pixel > 1, then the last three dimensions returned by aicsimage will be YXS.   
If samples per pixel ==1 then the S dimension will not be present and the last two dimensions will be YX.

Also, for AICS, a 3i reader is currently at least as high priority as a Zarr reader in the near term.

JacksonMaxfield, 2020-11-17:
Agree!

I think I was just including `Zarr` as potential for 4.0.0 release as it should be much easier to add than 3i since that is a custom reader.