AllenCellModeling/aicsimageio #171: is_pyramid -> multiscale

dimi-huer, 2020-12-11:
https://github.com/AllenCellModeling/aicsimageio/blob/03c20780089529a50f40c59968b8c0664137430e/aicsimageio/aics_image.py#L534

is_pyramid got replaced by multiscale=True/False

napari 0.3.0 (https://github.com/napari/napari/blob/5c23220938488c01b876183b4a488acca55a2c36/docs/release/release_0_3_0.md)

toloudis, 2020-12-12:
We intend to deprecate view_napari in favor of the napari-aicsimageio plugin.   We suggest you use that library moving forward.  Ultimately we expect to remove view_napari in aicsimageio 4.x.
@JacksonMaxfield may have more to add here.
https://github.com/AllenCellModeling/napari-aicsimageio

JacksonMaxfield, 2020-12-12:
Echoing @toloudis response that because we are actively working on 4.0 which deprecates this function, and, `napari-aicsimageio` is a better and more stable route for the same (if not better) functionality, I am going to put this at very low priority.

If you wanted to make a PR to update the function to be useable for the lifetime of 3.x I would happily review and accept however!

dimi-huer, 2020-12-13:
Thank you for the recommendation.

I know this part would not belong here, but since you suggested it here, I may ask something about napari-aicsimageio.
I believe there is more documentation about to come, because so far I am struggling with getting the same behavior in a script vs opening a ".czi" file in the GUI? Do I have to load the plugin specifically?

JacksonMaxfield, 2020-12-13:
Thanks for the PR!

Answering the programmatic usage of plugin. It _should_ be used as the default if `napari-aicsimageio` is installed and you run:
```python
import napari

with napari.gui_qt():
    # defaults to using aicsimageio-delayed if `napari-aicsimageio` is installed
    viewer = napari.view_image("my_file.ome.tiff")
```

But I do also know a `plugin` kwarg exists on certain objects / functions:
```python
import napari

with napari.gui_qt():
    viewer = napari.Viewer()
    viewer.open("path/to/file.czi")  # or (<- uses default plugin order)
    # viewer.open("path/to/file.czi", plugin="aicsimageio")  # or
    # viewer.open("path/to/file.czi", plugin="aicsimageio_delayed")
```

closed 2020-12-13 by JacksonMaxfield

JacksonMaxfield, 2020-12-13:
Oh! You may also be having the same issue that was just solved: https://github.com/AllenCellModeling/napari-aicsimageio/pull/10

I haven't released a new version of `napari-aicsimageio` yet so while it works on `master` there isn't a pypi release with the bugfix. Waiting on getting py39 build out (should be on Monday!)

JacksonMaxfield, 2020-12-14:
This should now be patched and released! `aicsimageio==3.3.3`.

A new `napari-aicsimageio` version to be released shortly as well. Hopefully one of the two or both resolve the issues and thanks again for the contribution!

PR #172 (2020-12-13): bugfix/replace-napari-is-pyramid-with-multiscale
## Description
I exchanged is_pyramid with multiscale.

Resolves #171 

I am sorry, but this is the first time for me doing this, so I am not sure on what kind of test files would be needed to attach. However, tests on py38 and lint succeeded with a coverage of 82%.