AllenCellModeling/aicsimageio #186: Inconsistent dimension ordering

zsdqui, 2021-01-24:
## System and Software
* aicsimageio Version: 3.0.7
* Python Version: 3.8.5
* Operating System: Ubuntu 18.04.5 LTS

## Description
*A clear description of the bug*
The default configurations of the OME-TIFF reader and writer are not consistent in dimension order. Reader reads "stczyx" and writer defaults to "stzcyx" which leads to unexpected behavior and errors when using argument "channel_names" in the OmeTIFF Writer module.

## Expected Behavior
*What did you expect to happen instead?*
I expect the order of dimensions to be preserved (this can currently be achieved by specifying dimension order in writer.OmeTiffWriter.save(..., dimension_order = "czyx" but should be default order to maintain consistency between read and write).

## Reproduction
*A minimal example that exhibits the behavior.*
`
import numpy as np
from aicsimageio import writers, AICSImage

data = np.random.rand(1,10,100,100) #intending czyx order based on how data is usually read
with writers.ome_tiff_writer.OmeTiffWriter("/home/user/data/test.ome.tiff", overwrite_file=True) as writer:
    writer.save(data)  #writer saves in zcyx order

data2 = AICSImage("/home/user/data/test.ome.tiff").get_image_data()
data2.shape #shape will be (1, 1, 10, 1, 100, 100) in stczyx order so c and z has been transposed
`

## Environment
*Any additional information about your environment*

JacksonMaxfield, 2021-01-24:
Hey @zsdqui!

Yep, we originally had a reason to do this but in hindsight, probably not the greatest decision 😅

You can get around this by including a `dimension_order` parameter to the `save` function.
```python
data = np.random.rand(1,10,100,100) # intending czyx order based on how data is usually read
with writers.ome_tiff_writer.OmeTiffWriter("/home/user/data/test.ome.tiff", overwrite_file=True) as writer:
    writer.save(data, dimension_order="CZYX")

data2 = AICSImage("/home/user/data/test.ome.tiff").get_image_data()
data2.shape   # shape will be correct
```

This will be _completely_ resolved in our 4.x release as well. All of the writers will accept data in the standard dimension order + have this same `dimension_order` param.

closed 2021-01-24 by JacksonMaxfield

JacksonMaxfield, 2021-01-24:
I will also offer a link to the 3.0.7 docstring of `OmeTiffWriter.save`: [docstring](https://github.com/AllenCellModeling/aicsimageio/blob/403a5065efdec8ad942b866bb11d815118e7acfe/aicsimageio/writers/ome_tiff_writer.py#L70)

toloudis, 2021-01-24:
For some more context, we were confused about what the "standard" dimension order _should_ be, back in the very early days of the library, and didn't want to make breaking changes to the writer's default behavior once we settled on the standard order  STCZYX.  @JacksonMaxfield @heeler this makes me think a short "porting guide" document would be nice for those who want to go from 3.x to 4.x!  Just something that names the breaking changes, or what parts of your code may require modification.