AllenCellModeling/aicsimageio #193: 4.0.0-dev0 aicspylibczi requirement fails to find matching distribution

bryantChhun, 2021-01-29:
## System and Software
* aicsimageio Version: 4.0.0-dev0
* Python Version: 3.7
* Operating System: OSX 10.13.6

## Description
During installation I get:

>chhun-mbp:~ bryant.chhun$ pip install acispylibczi==2.7.5
ERROR: Could not find a version that satisfies the requirement acispylibczi==2.7.5
ERROR: No matching distribution found for acispylibczi==2.7.5

As described here: https://github.com/AllenCellModeling/aicspylibczi/issues/68
aicspylibczi does not find matching distribution for mac OS 10.13.

This is fixed in aicspylibczi=2.8.0

## Expected Behavior
*What did you expect to happen instead?*

## Reproduction
*A minimal example that exhibits the behavior.*

## Environment
*Any additional information about your environment*

JacksonMaxfield, 2021-01-29:
:sweat_smile: My bad!

Will try to release a very fast `dev1` to fix this!

JacksonMaxfield, 2021-01-29:
`aicsimageio==4.0.0.dev1` released on pypi that patches this!

PR #194 (2021-01-29): admin/temp-bump-aicspylibcz
## Description

`aicspylibczi` needs to be bumped to support Mac OS variants. This is a temporary patch prior to bringing in `aicspylibczi==3.0.0`

## Pull request recommendations:
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-12], adds tiff file format support_

Resolves #193 

- [x] Provide relevant tests for your feature or bug fix.
- [x] Provide or update documentation for any feature added by your pull request.

Thanks for contributi [... cut]
closed 2021-01-29 by JacksonMaxfield