AllenCellModeling/aicsimageio #195: XMLSchemaValidationError when opening micro-manager 2.0-gamma data on 4.0.0-dev0

bryantChhun, 2021-01-29:
## System and Software
* aicsimageio Version: 4.0.0-dev0
* Python Version: 3.7
* Operating System: OSX 10.13.6

## Description
This could be a feature request or a bug.

When parsing ome.tif files generated by micro-manager 2.0 gamma, I receive this error:

```---------------------------------------------------------------------------
XMLSchemaValidationError                  Traceback (most recent call last)
<ipython-input-5-c7607dfa71ef> in <module>
     11 
     12 # img = AICSImage('/Users/bryant.chhun/Desktop/Data/reconstruct-order-2/image_stack_tpzc_50tp_4p_5z_3c_2k_1/image_files_tpzc_50tp_4p_5z_3c_2k_1_MMStack_2-Pos000_000.ome.tif')
---> 13 img = OmeTiffReader('/Users/bryant.chhun/Desktop/Data/reconstruct-order-2/image_stack_tpzc_50tp_2p_5z_3c_512k_1/image_stack_tpzc_50tp_2p_5z_3c_512k_1_MMStack_2-Pos000_000.ome.tif')

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/aicsimageio/readers/ome_tiff_reader.py in __init__(self, image, chunk_by_dims, clean_metadata)
     96 
     97         # Enforce valid image
---> 98         if not self._is_supported_image(self.fs, self.path, clean_metadata):
     99             raise exceptions.UnsupportedFileFormatError(
    100                 self.__class__.__name__, self.path

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/aicsimageio/readers/ome_tiff_reader.py in _is_supported_image(fs, path, clean_metadata, **kwargs)
     49                     # Get first page description (aka the description tag in general)
     50                     xml = tiff.pages[0].description
---> 51                     return OmeTiffReader._get_ome(xml, clean_metadata)
     52 
     53         # tifffile exception, tifffile exception, ome-types / etree exception

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/aicsimageio/readers/ome_tiff_reader.py in _get_ome(ome_xml, clean_metadata)
     38             ome_xml = metadata_utils.clean_ome_xml_for_known_issues(ome_xml)
     39 
---> 40         return from_xml(ome_xml)
     41 
     42     @staticmethod

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/ome_types/__init__.py in from_xml(xml)
     35     """
     36     xml = os.fspath(xml)
---> 37     d = to_dict(xml)
     38     for key in list(d.keys()):
     39         if key.startswith(("xml", "xsi")):

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/ome_types/schema.py in to_dict(xml, schema, converter, **kwargs)
    253 ) -> Dict[str, Any]:
    254     schema = schema or get_schema(xml)
--> 255     result = schema.to_dict(xml, converter=converter, **kwargs)
    256     # xmlschema doesn't provide usable access to mixed XML content, so we'll
    257     # fill the XMLAnnotation value attributes ourselves by re-parsing the XML

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/xmlschema/validators/schema.py in decode(self, source, path, schema_path, validation, *args, **kwargs)
   1684         """
   1685         data, errors = [], []
-> 1686         for result in self.iter_decode(source, path, schema_path, validation, *args, **kwargs):
   1687             if not isinstance(result, XMLSchemaValidationError):
   1688                 data.append(result)

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/xmlschema/validators/schema.py in iter_decode(self, source, path, schema_path, validation, process_namespaces, namespaces, use_defaults, decimal_type, datetime_types, binary_types, converter, filler, fill_missing, keep_unknown, max_depth, depth_filler, **kwargs)
   1674                     return
   1675 
-> 1676             yield from xsd_element.iter_decode(elem, validation, **kwargs)
   1677 
   1678         if 'max_depth' not in kwargs:

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/xmlschema/validators/elements.py in iter_decode(self, elem, validation, **kwargs)
    620                     yield self.validation_error(validation, error, **kwargs)
    621 
--> 622             for result in content_decoder.iter_decode(elem, validation, **kwargs):
    623                 if isinstance(result, XMLSchemaValidationError):
    624                     yield self.validation_error(validation, result, elem, **kwargs)

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/xmlschema/validators/groups.py in iter_decode(self, elem, validation, **kwargs)
    755                 continue
    756 
--> 757             for result in xsd_element.iter_decode(child, validation, **kwargs):
    758                 if isinstance(result, XMLSchemaValidationError):
    759                     yield result

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/xmlschema/validators/elements.py in iter_decode(self, elem, validation, **kwargs)
    577         for result in attribute_group.iter_decode(elem.attrib, validation, **kwargs):
    578             if isinstance(result, XMLSchemaValidationError):
--> 579                 yield self.validation_error(validation, result, elem, **kwargs)
    580             else:
    581                 attributes = result

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/xmlschema/validators/xsdbase.py in validation_error(self, validation, error, obj, source, namespaces, **_kwargs)
    884 
    885         if validation == 'strict' and error.elem is not None:
--> 886             raise error
    887         return error

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/xmlschema/validators/simple_types.py in iter_decode(self, obj, validation, **kwargs)
   1290             if not isinstance(self.primitive_type, XsdUnion):
   1291                 try:
-> 1292                     self.patterns(obj)
   1293                 except XMLSchemaValidationError as err:
   1294                     yield err

~/anaconda3/envs/reconorder2.0/lib/python3.7/site-packages/xmlschema/validators/facets.py in __call__(self, text)
    651             if all(pattern.match(text) is None for pattern in self.patterns):
    652                 reason = "value doesn't match any pattern of {!r}".format(self.regexps)
--> 653                 raise XMLSchemaValidationError(self, text, reason)
    654         except TypeError as err:
    655             raise XMLSchemaValidationError(self, text, str(err)) from None

XMLSchemaValidationError: failed validating 'Microscope' with XsdPatternFacets(['(urn:lsid:([\\w\\-\\.]+\\.[\\w\\-\\.]+)+:Instrument:\\S+)|(Instrumen...']):

Reason: attribute ID='Microscope': value doesn't match any pattern of ['(urn:lsid:([\\w\\-\\.]+\\.[\\w\\-\\.]+)+:Instrument:\\S+)|(Instrument:\\S+)']

Schema:

  <xsd:pattern xmlns:xsd="http://www.w3.org/2001/XMLSchema" value="(urn:lsid:([\w\-\.]+\.[\w\-\.]+)+:Instrument:\S+)|(Instrument:\S+)" />

Instance:

  <Instrument xmlns="http://www.openmicroscopy.org/Schemas/OME/2016-06" ID="Microscope"><Detector ID="Camera" Manufacturer="DCam" Model="DemoCamera-MultiMode" Offset="0.0" SerialNumber="V1.0" /></Instrument>

Path: /OME/Instrument
```

Sample data created using mm2-gamma 20201209 build:
[Google drive link](https://drive.google.com/drive/folders/1lglcWF9AUsqzZzVaUBzrlQO2TG9anPcf?usp=sharing
)

## Expected Behavior
It appears there is an issue with the mm2-gamma XML metadata structure.  This wasn't a problem in earlier versions of aicsimageio, but I understand 4.0.0-dev0 parses coordinates differently.

This brings up a broader question that our group is interested in, and could warrant a new issue.  We are interested in pushing micro-manager towards modern data formats that are compatible with modern python tools, such as zarr/dask/ometiff etc... I'd be happy to start a new reader specific to micro-manager files, built upon the interface/spec you define, if that is the appropriate procedure.

## Reproduction
The error is thrown when running the following code:
```
from aicsimageio import AICSImage
from aicsimageio.readers import OmeTiffReader

# either AICSImage
img = AICSImage(<path to data from google drive above>)

# or directly calling OmeTiffReader
img = OmeTiffReader(<path to data from google drive above>)

```

## Environment
*Any additional information about your environment*

JacksonMaxfield, 2021-01-29:
First, can you clarify that this error occurs with `AICSImage`?

What _should_ have happened is `AICSImage` tries to use `OmeTiffReader` and it fails / results in an error and rolls back to using `TiffReader`. But from your issue it looks like you are saying it raised an error `AICSImage` instead of rolling back.

Second, yes we changed the underlying metadata parser for `OmeTiffReader` from a _very_ outdated "vendor" module to [ome-types](https://github.com/tlambert03/ome-types) which validates that the metadata in the file is valid OME-XML. (Which is very useful for both usage and downstream API consistency).

It would probably be good to add a `MicroManagerReader` or similar to handle an extension of the OME model but I do want to make sure that the first issue is resolved.

JacksonMaxfield, 2021-01-29:
I just tried out the provided images myself and yes -- `AICSImage` fails for one of the files because the `XMLSchemaValidationError` exception wasn't caught. I have patched that and will push a branch later today. Interestingly, the other file will initialize but won't actually read the file and will take more investigation.

I will have an update later today this weekend on this.

Thanks for early testing btw! I really appreciate it!

bryantChhun, 2021-01-29:
No problem!  happy to give this a spin.  This repo is a well timed and represents a nice convergence of needs.
I could go ahead and start a new issue/feature request for a `MicroManagerReader` to discuss those needs and also to loop in micro-manager devs.  The metadata schema may be evolving there.

JacksonMaxfield, 2021-01-29:
I personally think that is the best strategy for moving forward. I doub't we will get a full `MicroManagerReader` into the 4.0 release unless the devs themselves contributed it but I will for sure try to patch out the above bug to make it so that it can at least be read by `TiffReader`.

cc @toloudis @heeler

JacksonMaxfield, 2021-01-29:
The other comment is that from the rough look of things we might not need a `MicroManagerReader` just that micromanager should patch its OMX XML creation process to have valid to the spec XML :shrug:

JacksonMaxfield, 2021-01-30:
@bryantChhun I posted what I think is the issue to micromanager: https://github.com/micro-manager/micro-manager/issues/1094

Maybe they will get out a patch and this will be resolved upstream!

JacksonMaxfield, 2021-01-31:
Alright @bryantChhun I have some updates:

I have a branch that will read MicroManager files like "normal" OME TIFFs.
Branch: [bugfix/catch-all-errors-in-ome-tiff](https://github.com/AllenCellModeling/aicsimageio/tree/bugfix/catch-all-errors-in-ome-tiff)

If the file is the "main" / "master" OME TIFF of the micromanager series, `aicsimageio` will utilize `OmeTiffReader` and all of the "attached" files besides the main one will act as scenes.

If the file is an "attached" OME TIFF of the micromanager series, `aicsimageio` will roll back to just `TiffReader` and will only have access to just that scene.

I.E.
```python
from aicsimageio import AICSImage

main_img = AICSImage("micromanager-main-img.ome.tiff")  # utilizes OmeTiffReader
main_img.scenes  # returns tuple of all the images available and you can proceed as normal
main_img.metadata  # returns the full OME object for use

attached_img = AICSImage("micromanager-attached-img.ome.tiff")  # utilizes TiffReader
attached_img.scenes  # returns tuple of just "Image:0", aka, just this file
attached_img.metadata  # returns None, as it doesn't pull the metadata from the parent file
```

I would say this is a temporary workaround for now as I think a longer term `MicroManagerReader` as defined / discussed in #196 would be a bit better for numerous reasons:

1. better metadata management
2. handling of remote files (currently, if you try to read the main file remotely but then try to switch scenes to access the contents of a different file it will fail because it thinks the attachment is local)

But at least we can support them in general.

I am cleaning up the metadata errors prior to reading it in as I would like to support these files without just saying: "please upgrade your micromanager version to something later than this PR: https://github.com/micro-manager/micro-manager/pull/1095 and re-aquire your dataset" haha so you should be all good to give this a try.

There is one issue I am still trying to debug... and uhhhh, it seems totally random?
I haven't made a PR yet because this branch / the specific tests for micromanager files fail (randomly) on delayed chunk reads.

I am not kidding. I can run:
```python
img = AICSImage("image_stack_tpzc_50tp_2p_5z_3c_512k_1_MMStack_2-Pos000_000.ome.tif")
img.set_scene("Image:1")
img.get_image_dask_data("YX").compute()
```

<summary>
And it will fail 80% of the time and succeed 20% of the time. The error that is raised when it fails is always the same:</summary>
<details>

```python
~/active/cell/tifffile/tifffile/tifffile.py in asarray(self, out, squeeze, lock, maxworkers)
   6101                 out = create_output(out, keyframe.shaped, keyframe._dtype)
   6102             with lock:
-> 6103                 fh.seek(self.dataoffsets[0])
   6104                 result = fh.read_array(
   6105                     keyframe.parent.byteorder + keyframe._dtype.char,

~/active/cell/tifffile/tifffile/tifffile.py in seek(self, offset, whence)
   8746                 self._fh.seek(self._offset + self._size + offset, 0)
   8747                 return
-> 8748         self._fh.seek(offset, whence)
   8749 
   8750     def read(self, size=-1):

ValueError: seek of closed file
```

</details>

So, while you can give the branch a go, probably don't try chunk reading until I patch that issue / figure out how to test it reliably.

cgohlke, 2021-02-02:
Re `ValueError: seek of closed file`: I think this can happen when multiple `ZarrTiffStore` instances are using the same `TiffFile` instance and are accessed concurrently from multiple threads. In this case the issue should be fixed by using the same `RLock` for instantiating the `ZarrTiffStore`.

JacksonMaxfield, 2021-02-02:
Hmmmm. I figured it might be something like that as well but I am a bit confused as to why all of the other tests with `TiffReader` would succeed and not show this behavior but it's just the micromanager / multi-file TIFFs that fail this behavior. Is anything special happening for the `ZarrTiffStore` for those types of files?

JacksonMaxfield, 2021-02-02:
Actually I think I just answered my own question in my head. I think this needs to be handled as a part of the `OmeTiffReader` specification for us. If an image utilizes this multi-file approach we need to handle the full file management part. :/

cgohlke, 2021-02-02:
I would expect other kind of problems (e.g. silent corruptions for uncompressed image data) also for single files. Maybe less often since seek and read are successive statements. Do you test that the data returned by get_image_dask_data match the data returned by tifffile asarray functions? The next tifffile version will use a RLock unique to a TiffFile instance, which might fix this issue.

JacksonMaxfield, 2021-02-02:
Well this isn't good...

[We do run tests for `get_image_dask_data` vs `get_image_data`](https://github.com/AllenCellModeling/aicsimageio/blob/main/aicsimageio/tests/image_container_test_utils.py#L81) but as you can see that test only checks the first YX plane. So just to be safe I locally just tried changing that test to getting the `ZYX` and got some failures. Not all failures, but some.

Are you saying the next tifffile release will patch the `RLock` behavior or the "silent corruptions" error?

JacksonMaxfield, 2021-02-02:
Actually hold on. I think this is a bug in my index selection management when I implemented the `chunk_by_dims` parameter.

JacksonMaxfield, 2021-02-02:
Alright. So I have a patch for the bug that was just found in this thread. I still don't have a patch for the "random failure issue" as described above.

Will make a PR for the bug tomorrow after benchmarks finish on my local machine

Will wait to work on finishing resolving of the specific issue after next tifffile release.

JacksonMaxfield, 2021-02-20:
Resolved in #207. Will release a dev2 tonight.

closed 2021-02-20 by JacksonMaxfield