AllenCellModeling/aicsimageio #203: UnsupportedFileFormatError when opening micromanager 2.0 gamma ome.tiff on 4.0-dev1@bugfix/catch-all-errors branch

bryantChhun, 2021-02-03:
## System and Software
* aicsimageio Version: 4.0-dev1, bugfix/catch-all-errors-in-ome-tiff
* Python Version: 3.7
* Operating System: OS 10.13.6

## Description
When attempting to open the ome tiff [attached here](https://drive.google.com/drive/folders/14LFIZ9FCq3CzYeKmVEB10BwvefF7AdDW?usp=sharing), I receive the following error:

[image]

I believe there is some aspect of the ome metadata here that is not caught by `clean_ome_xml_for_known_issues` ?

## Expected Behavior
*What did you expect to happen instead?*

## Reproduction
*A minimal example that exhibits the behavior.*

## Environment
*Any additional information about your environment*

JacksonMaxfield, 2021-02-03:
Yea I haven't worked on that branch in a bit. Been slammed with other work.

During dev meeting we discussed moving the log level from [`log.debug` to `log.error`](https://github.com/AllenCellModeling/aicsimageio/blob/bugfix/catch-all-errors-in-ome-tiff/aicsimageio/readers/ome_tiff_reader.py#L70) so that it will always be seen as to what went wrong during reading.

I am a bit slammed today / this week, any chance you could make some changes locally in that branch to further debug which metadata is failing validation.

JacksonMaxfield, 2021-02-03:
Oh, if you do have the opportunity to do so, my link to the code is meant that both `ET.parse` errors _and_ `XMLSchemaValidationErrors` would be raised from `log.debug` to `log.error`. So both of those except statements.

bryantChhun, 2021-02-03:
> I am a bit slammed today / this week, any chance you could make some changes locally in that branch to further debug which metadata is failing validation.

Totally understood.  I also have quite a bit of other work but can take a look when I have time.

Thanks!

bryantChhun, 2021-02-09:
It looks like it's another `XMLSchemaValidationError` with a particular field in the xml.  This is a slightly older dataset, but likely, it's more an issue of micro-manager not following specific XML standards.

Info on the error as reported by my IDE is here:
[image]

My current thought is to solve inconsistencies with micro-manager 2.0-gamma's current build (submitting PRs to micro-manager), then work backwards through older versions (patching `metadata.utils.clean_ome_xml_for_known_issues` here to accommodate older builds).

JacksonMaxfield, 2021-02-09:
Yea these should be much easier to bebug when we release dev2. Will try to get this out this week. Apologies on the delay.

Likely what we will want to do is split up that giant metadata cleaning function into parts...

`clean_ome_xml_for_known_issues` should call out to `clean_for_aicsimageio_3_issues` and `clean_micromanager_issues` and etc.

I somewhat expect that single function to be the most "community contributed" function which is both sad and exciting. Hopefully we can head towards better metadata practices through schema validation.

bryantChhun, 2021-02-09:
> Hopefully we can head towards better metadata practices through schema validation.

Speaking of schema validation, this is unexplored territory for me.  If tools don't exist to compare vs a schema, reporting differences in an intelligible way, we should make one :-)

JacksonMaxfield, 2021-02-09:
That is a good tool and, `xmlschema` the library we use to validate actually has _great_ error / reports. Its just that we hide them away from you as a user on `dev0` and `dev1`... `dev2` will raise the error to the user. The errors are formatted similar to the bioformats tool.

> Instrument elements must have an ID attribute:
> Received: ....
> Must follow pattern: ...

etc.

JacksonMaxfield, 2021-02-16:
Finally back! and with updates:

`xmlschema` validation reporting the same issue as xmlvalidate tool:

```python
In [2]: img = AICSImage("/home/maxfield/Downloads/20210114_5_state_2_1_MMStack.ome.tif")
[ERROR: ome_tiff_reader:  77 2021-02-15 16:10:53,939] OME XML validation failed
[ERROR: ome_tiff_reader:  78 2021-02-15 16:10:53,939] failed validating 'HamamatsuHam_DCAM' with XsdPatternFacets(['(urn:lsid:([\\w\\-\\.]+\\.[\\w\\-\\.]+)+:Detector:\\S+)|(Detector:\\...']):

Reason: attribute ID='HamamatsuHam_DCAM': value doesn't match any pattern of ['(urn:lsid:([\\w\\-\\.]+\\.[\\w\\-\\.]+)+:Detector:\\S+)|(Detector:\\S+)']

Schema:

  <xsd:pattern xmlns:xsd="http://www.w3.org/2001/XMLSchema" value="(urn:lsid:([\w\-\.]+\.[\w\-\.]+)+:Detector:\S+)|(Detector:\S+)" />

Instance:

  <Detector xmlns="http://www.openmicroscopy.org/Schemas/OME/2016-06" ID="HamamatsuHam_DCAM" Manufacturer="HamamatsuHam_DCAM" Model="C11440-42U" SerialNumber="S/N: 000382" />

Path: /OME/Instrument/Detector
```

I am about to push a commit that changes detector name cleaning to:
```python
for detector_index, detector in enumerate(
    instrument.findall(f"{namespace}Detector")
):
    detector_id = detector.get("ID")

    # Create ome detector id if needed
    ome_detector_id = None
    if detector_id == "Camera":
        ome_detector_id = generate_ome_detector_id(detector_index)
    elif not detector_id.startswith("Detector:"):
        ome_detector_id = generate_ome_detector_id(detector_id)

    # Apply ome detector id if replaced
    if ome_detector_id is not None:
        detector.set("ID", ome_detector_id)
        metadata_changes.append(
            f"Updated attribute 'ID' from '{detector_id}' to "
            f"'{ome_detector_id}' on Detector element at "
            f"position {detector_index}."
        )
```

Which basically says "look if it's the default, 'Camera'" replace with something else entirely, if it has a name in general, prepend it with "Detector:". Resolved all the issues entirely and I can now do:

```python
In [1]: from aicsimageio import AICSImage
[WARNING: registry: 485 2021-02-15 16:19:14,265] Redefining 'li' (<class 'pint.definitions.UnitDefinition'>)
[INFO] Resource 'XMLSchema.xsd' is already loaded
[INFO: schema:1026 2021-02-15 16:19:14,474] Resource 'XMLSchema.xsd' is already loaded

In [2]: img = AICSImage("/home/maxfield/Downloads/20210114_5_state_2_1_MMStack.ome.tif")

In [3]: img
Out[3]: <AICSImage [Reader: OmeTiffReader, Image-is-in-Memory: False]>
```

So looks good to me!

JacksonMaxfield, 2021-02-20:
Resolved in #207. Will release a dev2 tonight.

closed 2021-02-20 by JacksonMaxfield

JacksonMaxfield, 2021-02-20:
@bryantChhun dev2 has been released which includes the tiff chunk reading fix and the various ome tiff reader xml parsing fixes.