AllenCellModeling/aicsimageio #205: Add tests for remote writing

JacksonMaxfield, 2021-02-15:
In #198 we added back writing functionality to the 4.0 branch but we haven't yet decided how to handle remote writing so this issue will be used for tracking that so I can merge that PR.

Currently implimentations with `moto` do not work because `fsspec` uses `aiobotocore` instead of base `botocore` which results in a mocking error.

`fsspec` / `s3fs` mocks this by creating a temporary s3 bucket that lives for the lifetime of the tests.

cc @anl @toloudis

JacksonMaxfield, 2021-03-24:
Should be done whenever we add an `OmeZarrWriter` as defined in #192

closed 2023-03-20 by evamaxfield

toloudis, 2023-03-20:
Just as a note: remote writing in automated unit tests is a no-go due to reasons of cost and safety.