AllenCellModeling/aicsimageio #206: Check that size of channels dim on is corrected for samples in OmeTiffReader

JacksonMaxfield, 2021-02-15:
On `main` / `4.x` branch, we utilize the OME object and specifically we get dimension sizes using `size_{dim}` but that size isn't actually the number of channels, it is the number of channel elements * samples per pixel which could lead to problems for RGB multi-channel OME Tiff Images.

We can correct this by simply dividing that value by the samples per pixel.

toloudis, 2021-02-15:
also note that samples per pixel is a per-Channel attribute!  num channels is len(ome_obj.image[i].pixels.channels)

JacksonMaxfield, 2021-02-15:
> also note that samples per pixel is a per-Channel attribute! num channels is len(ome_obj.image[i].pixels.channels)

I don't know if we have ever encountered an image with varying num samples per "Image" / Scene though.... / is that possible?

PR #211 (2021-03-04): feature/ome-tiff-writer-4
OmeTiffWriter
- handles rgb data
- handles multi-scene writing
- does a whole bunch of argument validation.

Basic usage:
OmeTiffWriter.save(data_array or list of data_arrays, filename, metadata...)

if no metadata is provided, some will be auto-generated.

if you pass in metadata piecemeal,( dimension order,channel names etc) then we will build that into the generated ome metadata for you.

if you pass in your own ome metadata, we will validate it and do a couple of basic dimension consistency checks.

if you want to pass in RGB data, you have to tell the writer by p [... cut]
closed 2021-03-04 by JacksonMaxfield