AllenCellModeling/aicsimageio #209: Application crashes when trying to read the shape of a CZI file with WinError 10054

sebi06, 2021-02-18:
Hi all,

I use the latest AICSImageIO + aicspylibczi on a WIn10, Python 3.7.4

I have a CZI file, which works fine in ZEN and also in Fiji etc. No I try to do the following

```python
czi_aics = AICSImage(filename)
metadata['Axes_aics'] = czi_aics.dims
try:
    metadata['Shape_aics'] = czi_aics.shape
    ....
```
The 1st and 2nd line work OK. It shows returns `'STCZYX'`, which is fine. But when I move on the call `czi_aics.shape` terminates the whole script and the only error message I can see is ( when trying to execute this line in the debugger):

```text
metadata['Shape_aics'] = czi_aics.shape
[WinError 10054] An existing connection was forcibly closed by the remote host
```

Has anyone an idea, what is going on? Important: The same code works fine for most CZIs, but not for at lest 2 I have. But from I all I can say, they are totally fine.

sebi06, 2021-02-18:
Here is the CZI file:

https://www.dropbox.com/s/82zohgg2ivom7mq/czi_crash.czi?dl=0

that does not work.

heeler, 2021-02-18:
Thanks @sebi06! I'll have a look ASAP.

JacksonMaxfield, 2021-03-08:
Sorry for the delay in getting to this @sebi06. We are working on 4.0 stuff and will return to this as soon as possible. It is on our radar so we will get back when we can.

Any updates or discoveries on your side?

sebi06, 2021-03-21:
Hi,

so far not. But I was also just not having enough time to dig deeper.

JacksonMaxfield, 2021-07-12:
@sebi06 just wanted to check up on this. With recent changes and patches to aicspylibczi and aicsimageio is this still a bug for you?

closed 2023-04-15 by github-actions[bot]