AllenCellModeling/aicsimageio #218: won't load multi-channel, mutli-scene czi file (windows 10)

cfrick13, 2021-03-12:
## System and Software
* aicsimageio Version: 4.0.0.dev3
* Python Version: Python 3.9
* Operating System: Windows 10 (v. 1909)

## Description
I try to load a standard multi-channel, mutli-scene `.czi ` file and it fails with the error "AICSImage does not support the image: "

## Expected Behavior
Image is recognized

## Reproduction
```
from aicsimageio import AICSImage
file_path = '20200323_F01_001.czi'
img = AICSImage(file_path)  # selects the first scene found
```
```
---------------------------------------------------------------------------
UnsupportedFileFormatError                Traceback (most recent call last)
<ipython-input-1-dec872bc3104> in <module>
      1 from aicsimageio import AICSImage
      2 file_path = '20200323_F01_001.czi'
----> 3 img = AICSImage(file_path,)  # selects the first scene found

c:\users\chris.frick\appdata\local\continuum\anaconda3\envs\new_aics\lib\site-packages\aicsimageio\aics_image.py in __init__(self, image, **kwargs)
    114         """
    115         # Determine reader class and create dask delayed array
--> 116         ReaderClass = self.determine_reader(image, **kwargs)
    117         self._reader = ReaderClass(image, **kwargs)
    118 

c:\users\chris.frick\appdata\local\continuum\anaconda3\envs\new_aics\lib\site-packages\aicsimageio\aics_image.py in determine_reader(image, **kwargs)
     59             path = str(type(image))
     60 
---> 61         raise exceptions.UnsupportedFileFormatError("AICSImage", path)
     62 
     63     def __init__(self, image: types.ImageLike, **kwargs: Any):

UnsupportedFileFormatError: AICSImage does not support the image: '20200323_F01_001.czi'.
```

## Environment
This was a freshly made environment in python 3.9. 
I ran the code above in a jupyter-notebook

JacksonMaxfield, 2021-03-12:
!!! Sorry should have warned. `CZIReader` is the only reader not yet in 4.0. @heeler is working on getting it in as soon as possible.

4.0.0.dev3 has readers for:
* OME-TIFF
* TIFF
* LIF
* the easy image types like PNG, JPEG, GIF, etc.

cfrick13, 2021-03-12:
Aha. Well that explains it. Thanks Jackson!
Also would be good to update this page with this info. https://allencellmodeling.github.io/aicsimageio/ (that is what made me think it was ready for CZI)

JacksonMaxfield, 2021-03-12:
Yea. I unfortunately think that is the only part of the docs that is out of sync.

JacksonMaxfield, 2021-03-12:
This will be closed as soon as `CZIReader` is done. Should be soonish.

PR #231 (2021-05-31): feature/czi-reader
## Description
This implements the 4.0 support for Zeiss Czi Files in the class CziReader.
For 4.0 local file reading will only be supported, I hope to add remote reading for 4.1. 
lxml has been replaced with XML for serialization. 
Mosaic file stitching is in. 

There are tests for reading M, S(or A at the lower level) and for retrieving only part of the required chunk dimensions (X, Y, S)=( : , : , 0 ).

Resolves #218 
Resolves #163 
Resolves #141 
Resolves #62 
Resolves #239
closed 2021-05-31 by JacksonMaxfield