AllenCellModeling/aicsimageio #219: make aicspylibczi an optional dependency

bryantChhun, 2021-03-15:
## Use Case
From what I understand `aicspylibczi` is a compiled library for reading CZI image files.  

The problem is when one needs to use this reader for other data types (.tif, .ome-tiff) on systems not supported by `aicspylibczi`.  Our organization has a mixture of OS systems (linux, OSX, IBM-Power9).  Importantly, IBM is [not supported by Github Actions.](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners#supported-runners-and-hardware-resources)

## Solution
While I think we can specify sub-dependencies in our setup.py, `aicspylibczi` is very specific to one file type that is not needed for any general non-zeiss user.  Perhaps an install pattern with explicitly defined backends, [like napari's](https://napari.org/tutorials/fundamentals/installation#choosing-a-different-qt-backend), would be nice: https://github.com/napari/napari/blob/ce9dd0692d5e6e7564ba44eb7a5a28e77828eba4/setup.cfg#L71

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*

JacksonMaxfield, 2021-03-15:
Hey @bryantChhun as soon as you posted this we pretty much instantly started to discuss this in our Slack. @heeler should be posting a workaround for now soon. But we are still discussing how to handle this. Probably won't have a response for you for a couple of days. Maybe next monday / prior to 4.0 release. Will keep you updated as best I can.

heeler, 2021-03-15:
your best immediate fix is 
`git clone --recurse-submodules https://github.com/AllenCellModeling/aicspylibczi.git`
`cd aicspylibczi`
`pip install .` 
that should build it locally (in the GH Action I actually use `pip install .[test]`) 

Try that and see if it works for local build. Compiling assumes g++ or clang++.

JacksonMaxfield, 2021-03-24:
#221 has been merged. Will release a new dev release shortly which should solve this for you @bryantChhun

closed 2021-03-24 by JacksonMaxfield

JacksonMaxfield, 2021-03-24:
https://pypi.org/project/aicsimageio/4.0.0.dev4/

released! Let me know if something doesn't work.

bryantChhun, 2021-03-24:
Wow thanks guys.  Nice work.
:-)

AetherUnbound, 2021-03-24:
Happy to help! We appreciate you bringing this up, we think this is a very forward-thinking change for the integration of more image formats in the future 🙂
Also, gentle reminder that not all of the members of the dev team for this project are guys: http://heyguys.cc/.

PR #221 (2021-03-24): admin/split-reader-deps
**This PR points at the doc updates branch.**

## Description

A decent implementation for handling splitting reader dependencies out. 

This makes `aicsimageio>=4.0.0` support TIFF + OME-TIFF + array-like by default but then anything else is an extra install.
Moving forward, the recommendation was to default support formats that OME supports. So OME-TIFF, OME-Zarr, etc.

When a format is **naively** detected as "supported" but backing lib not installed, an error will print out with the exact pip command to install the extra lib. I say naively because it works off of fi [... cut]