AllenCellModeling/aicsimageio #220: Issue with opening CZI file

niveditasa, 2021-03-15:
## System and Software
* aicsimageio Version:3.3.4
* Python Version:3.7.10
* Operating System:Windows

## Description
While using AICS image io to open and view one of the CZI files, I received the following error:
My code: 
reader = AICSImage(FILE_NAME) 
IMG = reader.data.astype(np.float32)
 
Error:
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
~\Anaconda3\envs\segmentation\lib\site-packages\aicsimageio\readers\reader.py in data(self)
    176                 # https://github.com/AllenCellModeling/aicsimageio/issues/82
--> 177                 sys.modules["distributed"].get_client()
    178 

KeyError: 'distributed'

During handling of the above exception, another exception occurred:
MemoryError                               Traceback (most recent call last)
<ipython-input-17-9392a512deeb> in <module>
      1 FILE_NAME =  r'path to my file name and file name'
      2 reader = AICSImage(FILE_NAME)
----> 3 IMG = reader.data.astype(np.float32)

~\Anaconda3\envs\segmentation\lib\site-packages\aicsimageio\aics_image.py in data(self)
    195         if self._data is None:
    196             self._data = transforms.reshape_data(
--> 197                 data=self.reader.data,
    198                 given_dims=self._known_dims or self.reader.dims,
    199                 return_dims=self.dims,

~\Anaconda3\envs\segmentation\lib\site-packages\aicsimageio\readers\reader.py in data(self)
    183                 self._data = self._dask_data.compute()
    184             except (KeyError, ValueError):
--> 185                 self._data = self._read_immediate()
    186                 self._dask_data = da.from_array(self._data)
    187 

~\Anaconda3\envs\segmentation\lib\site-packages\aicsimageio\readers\czi_reader.py in _read_immediate(self)
    445             czi.reader = None
    446 
--> 447             raise e
    448 
    449         return data

~\Anaconda3\envs\segmentation\lib\site-packages\aicsimageio\readers\czi_reader.py in _read_immediate(self)
    434                 # have the same dimensions
    435                 # Read all data in the image
--> 436                 data, _ = czi.read_image()
    437 
    438                 # A really bad way to close any connection to the CZI object

~\Anaconda3\envs\segmentation\lib\site-packages\aicspylibczi\CziFile.py in read_image(self, **kwargs)
    384         cores = self._get_cores_from_kwargs(kwargs)
    385 
--> 386         image, shape = self.reader.read_selected(plane_constraints, m_index, cores)
    387         return image, shape
    388 

MemoryError: bad allocation

## Expected Behavior
*What did you expect to happen instead?*

It should show no error and followed by print(IMG.shape) should print the Scene, Channels, Z,Y,X.

## Environment
conda env:segmentation

JacksonMaxfield, 2021-03-15:
`MemoryError: bad allocation` means that when numpy tried to allocate the space in RAM required for reading the image, it ran out of space.

This is exactly why `dask` is useful though!

You should try with the `get_image_dask_data` function:

```python
img = AICSImage("my-file.czi")
chunk = img.get_image_dask_data("ZYX", C=0, T=0, S=0)
chunk.compute()  # this will pull ONLY that chunk as a numpy array
```

You can then set the dimension to what you want in the function. I.e. if you want `T=1`:
`img.get_image_dask_data("ZYX", T=1)`

JacksonMaxfield, 2021-03-17:
Closing as issue seems to be resolved.

Dask delayed is great!

closed 2021-03-17 by JacksonMaxfield