AllenCellModeling/aicsimageio #222: 3i Slidebook Reader

JacksonMaxfield, 2021-03-24:
## Use Case
*Please provide a use case to help us understand your request in context*
AICS gene editing using 3i scopes for some imaging. These scopes produce `.sld` or slidebook files. Would be nice to be able to read all of AICS produced imaging data under one library.

## Solution
*Please describe your ideal solution*
I think @heeler talked about writing the module in C++ just like aicspylibczi. But we should first look around for other libraries that can read 3i. If there are none, I say we should take this as an opportunity to try out rust development.

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*

toloudis, 2021-03-24:
Some info / clues to reference implementations:
https://forum.image.sc/t/sld-bio-formats-errors/48353

jni, 2021-09-14:
Hi all, has anyone made any progress here, that might not be pushed to the latest `main`?

toloudis, 2021-09-14:
We don't have a complete implementation but there are some plans in motion to arrive at a working (maybe non optimal) solution, and some later plans to get a more optimal solution.

JacksonMaxfield, 2021-09-29:
@toloudis can we try to get somone from AICS to try 4.2 with a SLD file? Would love to close this.

JacksonMaxfield, 2021-09-29:
Actually I will tag @cfrick13. If you get a chance can you test a slidebook file?

cfrick13, 2021-09-29:
I tried multiple slidebook files and they currently do not work with the aicsimageio==4.2.0

JacksonMaxfield, 2021-09-29:
> I tried multiple slidebook files and they currently do not work with the aicsimageio==4.2.0

Mind sending us the files in dropbox or similar?

cfrick13, 2021-09-29:
```
from aicsimageio.readers import bioformats_reader

bf_reader = bioformats_reader.BioformatsReader(r"20210428_T02_002.sld")

```

```
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\_jpype.cp38-win_amd64.pyd in loci.formats.ImageReader.setId()

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\_jpype.cp38-win_amd64.pyd in loci.formats.FormatReader.setId()

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\_jpype.cp38-win_amd64.pyd in loci.formats.in.SlidebookReader.initFile()

Exception: Java Exception

The above exception was the direct cause of the following exception:

java.lang.ArrayIndexOutOfBoundsException  Traceback (most recent call last)
~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\aicsimageio\readers\bioformats_reader.py in __init__(self, image, original_meta, memoize, options)
    108         try:
--> 109             with BioFile(self._path, **self._bf_kwargs) as rdr:  # type: ignore
    110                 md = rdr._r.getMetadataStore()

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\aicsimageio\readers\bioformats_reader.py in __init__(self, path, series, meta, original_meta, memoize, options)
    279 
--> 280         self.open()
    281         self._lock = Lock()

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\aicsimageio\readers\bioformats_reader.py in open(self)
    308         """Open file."""
--> 309         self._r.setId(self._path)
    310 

java.lang.ArrayIndexOutOfBoundsException: java.lang.ArrayIndexOutOfBoundsException: Index 0 out of bounds for length 0

During handling of the above exception, another exception occurred:

UnsupportedFileFormatError                Traceback (most recent call last)
C:\Users\CHRIS~1.FRI\AppData\Local\Temp/ipykernel_17988/2797730550.py in <module>
      2 # 'sld': ['aicsimageio.readers.bioformats_reader.BioformatsReader'],
      3 
----> 4 bf_reader = bioformats_reader.BioformatsReader(r"\\allen\aics\assay-dev\MicroscopyData\John Paul\2021\20210428\20210428_T02_002.sld")

~\AppData\Local\Continuum\anaconda3\envs\aics_bf2\lib\site-packages\aicsimageio\readers\bioformats_reader.py in __init__(self, image, original_meta, memoize, options)
    115             raise
    116         except Exception:
--> 117             raise exceptions.UnsupportedFileFormatError(
    118                 self.__class__.__name__, self._path
    119             )

UnsupportedFileFormatError: BioformatsReader does not support the image: '20210428_T02_002.sld'.
```

tlambert03, 2021-09-29:
yeah, i'm not surprised to hear this.  3i took control of their own bioformats plugin, so it's not included with the bioformats_jar.  we'll need to handle this case specially (and I'm not sure what the deal with distribution of 3i's jar is)

tlambert03, 2021-09-29:
> Mind sending us the files in dropbox or similar?

but, it would be handy to have a sample!

cfrick13, 2021-09-29:
I slacked a .sld file to dan and jackson (faster than me trying to clean out my old dropbox). Hope that works. 
Excited to see .sld support soon hopefully :)

joshmoore, 2021-09-29:
> I'm not sure what the deal with distribution of 3i's jar is

see https://www.intelligent-imaging.com/technical-answers and http://sites.imagej.net/SlideBook/

tlambert03, 2021-09-29:
@joshmoore ... don't see any clear license info there.  do you have any insight on that?

joshmoore, 2021-09-29:
cc: @RichardMyers (https://github.com/ome/bioformats/pull/2289)

SeanLeRoy, 2023-03-21:
Closing this as duplicate of this slightly more recent issue #428

closed 2023-03-21 by SeanLeRoy