AllenCellModeling/aicsimageio #225: Reading metadata

miguel-fc, 2021-03-30:
Hi,

I'm interested in reading the metadata stored in some CZI files.

When I view these files with Fiji, and choose to display the metadata on OME-XML format, I can see there's a lot of information in the files. 

From the manual of aicsimageio, I understand the following commands are possible:

metadata= img.metadata 
img.get_channel_names(metadata)
img.get_physical_pixel_size(metadata)

I have tried them and they do work. But, are there commands to extra more information from the files? (See for example a snapshot of some of the information contained on one of the files.)

Thanks

[image]

ssomnath, 2021-03-30:
Hello, I am working with @miguel-fc on extracting the metadata. Is there a way to get the data out of the lxml.etree.ElementTree objects into something more conventional like a python dictionary? 
For example,
Can we get:
```
{"Filter: 1":
       {"TransmittanceRange":
                 {"CutIn": 450.0,
                   "CutInUnit": "nm",
                   "CutOut": 490.0,
                   "CutOutUnit": "nm"}
          }
}
```

?

heeler, 2021-03-30:
Hi, 
you should be able to get the full raw metadata out using the `.metadata` property. 
For a CZI file this will still be in the CZI XML format but you can use XPath to interrogate it to your heart's content. 
(It will either be returned as an etree or as a string, if it's a string I think the function is fromString to get it into an etree)
That should let you get any value you like that was written into the metadata.

As to OME-XML, currently we are working on mappings but they are not complete yet. The mappings in existence now, bioformats or other, only map a very limited portion of the metadata over. I state this just as a caution. 

I hope this helps!

JacksonMaxfield, 2021-03-30:
Hey @miguel-fc and @ssomnath, if you don't like XPath for exploring XML you can use `xmltodict` which is a package which does exactly what you asked from the looks of it.

https://github.com/martinblech/xmltodict

I haven't used it but that looks like the right direction.

Also I don't _think_ you should have to provide the `metadata` back to the `get_channel_names` function. It should just be:
```python
img = AICSImage(...)
all_metadata = img.metadata
channel_names = img.get_channel_names()  # if you have a multi scene file, you can provide the scene id here to get the channel names for a different scene.
```

https://github.com/AllenCellModeling/aicsimageio/blob/master/aicsimageio/aics_image.py#L587

I will also say that we are hoping to release aicsimageio 4.0.0 soon ish and while that won't introduce any new shorthands to key pieces of the metadata (i.e. `img.metadata`, `img.channel_names`, etc.) it will change the access pattern from function calls to simple properties.

toloudis, 2021-03-30:
Just one more high level comment, and to echo much of what @heeler said, because we realize how important metadata is.

aicsimageio is not designed to understand the complete metadata of the supported proprietary file formats **just yet**.  

As @heeler mentioned, we are working toward some automatic conversion from proprietary file format metadata to OME metadata but it's still in early stages.  Once this framework is ready (and CZI is likely to be the first format that does this) then you will be able to get a ome-types object to use.

So for now, you have to ask for the metadata from aicsimageio readers and parse it yourself.  You can also try bioformats in the meantime, to convert metadata to OME format.

JacksonMaxfield, 2021-04-14:
Wanted to see if you had other questions or comments @miguel-fc or @ssomnath. If not can I close this?

miguel-fc, 2021-04-14:
Yes. Sorry for the delay in closing this. Thanks for all your help!

closed 2021-04-14 by miguel-fc