AllenCellModeling/aicsimageio #239: TypeError: _io.BufferedReader when opening .czi file

rcasero, 2021-05-27:
## System and Software
* aicsimageio Version: aicsimageio-4.0.0.dev6
* Python Version: 3.8
* Operating System: ubuntu linux 20.04

## Description
Trying to open .czi file fails.

## Expected Behavior
That the file is opened.

## Reproduction
Install aicsimageio in local conda environment with python 3.8 running

`pip install "aicsimageio[czi] @ git+https://github.com/AllenCellModeling/aicsimageio.git@v4.0.0.dev6"`

Then try to open a .czi file

```
import aicsimageio

histo_file = '/home/rcasero/coxgroup_zeiss_test/___00000022___00000436_02052021-TBX15del964-pWAT-0028.czi'
im = aicsimageio.AICSImage(histo_file)
```

This returns the exception

```
>>> im = aicsimageio.AICSImage(histo_file)
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/rcasero/Software/miniconda3/envs/cytometer_zeiss/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 197, in __init__
    ReaderClass = self.determine_reader(image, **kwargs)
  File "/home/rcasero/Software/miniconda3/envs/cytometer_zeiss/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 146, in determine_reader
    if ReaderClass.is_supported_image(image, **kwargs):  # type: ignore
  File "/home/rcasero/Software/miniconda3/envs/cytometer_zeiss/lib/python3.8/site-packages/aicsimageio/readers/reader.py", line 100, in is_supported_image
    return cls._is_supported_image(fs, path, **kwargs)
  File "/home/rcasero/Software/miniconda3/envs/cytometer_zeiss/lib/python3.8/site-packages/aicsimageio/readers/lif_reader.py", line 62, in _is_supported_image
    LifFile(open_resource)
  File "/home/rcasero/Software/miniconda3/envs/cytometer_zeiss/lib/python3.8/site-packages/readlif/reader.py", line 427, in __init__
    f = open(filename, "rb")
TypeError: expected str, bytes or os.PathLike object, not _io.BufferedReader
```

## Environment
The file lives on a CIFS network share.

rcasero, 2021-05-27:
Um, there's a warning when trying to install czi support from the development head

```
$ pip install "aicsimageio[czi] @ git+https://github.com/AllenCellModeling/aicsimageio.git@v4.0.0.dev6"
Collecting aicsimageio[czi]@ git+https://github.com/AllenCellModeling/aicsimageio.git@v4.0.0.dev6
  Cloning https://github.com/AllenCellModeling/aicsimageio.git (to revision v4.0.0.dev6) to /tmp/pip-install-hq5biaen/aicsimageio_8ab2e491eccd488c901108f66c9ef1a0
  Running command git clone -q https://github.com/AllenCellModeling/aicsimageio.git /tmp/pip-install-hq5biaen/aicsimageio_8ab2e491eccd488c901108f66c9ef1a0
  Running command git checkout -q 46d66900e80a511568a2d716ec4528f37b385c83
WARNING: aicsimageio 4.0.0.dev6 does not provide the extra 'czi'
```

JacksonMaxfield, 2021-05-27:
Hey @rcasero yep... This is somewhat expected as CZI Reading is still in review for 4.0: https://github.com/AllenCellModeling/aicsimageio/pull/231

I got some work done it last night and all the major pieces are there but if need it now, I would recommend installing from that branch.

JacksonMaxfield, 2021-05-27:
I am a bit concerned about the error you got above.... that seems like it tried and failed with `LifReader` which is expected... just the error you received should have been different... It should have been a "UnsupportedFileFormat" error that we catch and move on to the next reader to try. I will look into this in a bit.

JacksonMaxfield, 2021-05-27:
I can't reproduce the weird TypeError above.. I am hoping it was just a "corrupted" env:

Here is my attempt to reproduce with a branch new env and it produces the "correct" behavior:
```bash
conda create --name test python=3.8 -y
conda activate test
pip install --pre aicsimageio
python
```

```python
>>> from aicsimageio import AICSImage
>>> img = AICSImage("active/cell/aicsimageio/aicsimageio/tests/resources/s_1_t_1_c_1_z_1.czi")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/maxfield/miniconda3/envs/test/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 197, in __init__
    ReaderClass = self.determine_reader(image, **kwargs)
  File "/home/maxfield/miniconda3/envs/test/lib/python3.8/site-packages/aicsimageio/aics_image.py", line 162, in determine_reader
    raise exceptions.UnsupportedFileFormatError(
aicsimageio.exceptions.UnsupportedFileFormatError: AICSImage does not support the image: 'active/cell/aicsimageio/aicsimageio/tests/resources/s_1_t_1_c_1_z_1.czi'. File extension suggests format: 'czi'. Install extra format dependency with: `pip install aicsimageio[czi]`. See all known format extensions and their extra install name with `aicsimageio.formats.FORMAT_IMPLEMENTATIONS`.
```

So I think the only issue here is that CZI reading just hasn't been merged yet.

rcasero, 2021-05-27:
Hi @JacksonMaxfield , thanks for your reply. I tried with bioformats and czifile, and also got errors. I installed a VM with Windows so that I could install Zeiss ZEN lite, and that also gave errors, so now I'm thinking that maybe the files are corrupt. I've asked the person who produced the images, and will get back to you when I know more.

JacksonMaxfield, 2021-05-28:
Ahhhhh I see. Going to close this for now because this issue is largely related to "CZI support not in the dev releases yet" but if you have an update to share in the future about "weird CZI files" we can probably try to take a look.

closed 2021-05-28 by JacksonMaxfield

JacksonMaxfield, 2021-05-31:
4.0.0.dev7 has been released if you wanted to give reading the file another go. Unless you found out that it was a corrupted file or something.

PR #231 (2021-05-31): feature/czi-reader
## Description
This implements the 4.0 support for Zeiss Czi Files in the class CziReader.
For 4.0 local file reading will only be supported, I hope to add remote reading for 4.1. 
lxml has been replaced with XML for serialization. 
Mosaic file stitching is in. 

There are tests for reading M, S(or A at the lower level) and for retrieving only part of the required chunk dimensions (X, Y, S)=( : , : , 0 ).

Resolves #218 
Resolves #163 
Resolves #141 
Resolves #62 
Resolves #239