AllenCellModeling/aicsimageio #243: Reading Tiff Files with Zero Description Offset

MosGeo, 2021-05-30:
## System and Software
* aicsimageio Version: 3.3.6 
* Python Version: 3.8
* Operating System: Windows 10

## Description
Some tiff files do not load. Specifically tif files generated by Zen Ziess export feature. It looks like they have a "description_offset"  equal to 0.

The offending line of code is line 337 in the tiff_reader.py and modifying it from **_return bytearray("")_** to **_return bytearray("", "utf8")_** fixes this issue.

## Expected Behavior
Reading the tiff file normally.

## Reproduction
1. Export czi file to tiff in Zen Zeiss.
2. Attempt to load the file using aicsimageio

## Environment
None

JacksonMaxfield, 2021-05-30:
Hey @MosGeo I personally don't have access to Zeiss Zen but it sounds like you are working on solving the issue. Would you mind making a PR to our `v3` branch?

MosGeo, 2021-05-31:
I will make one. This should be a simple and easy fix for me to start with :)

JacksonMaxfield, 2021-05-31:
Thanks for the patch! Resolved and released v3.3.7!

PR #245 (2021-05-31): bugfix/fix-tiff-no-description-offset
This is a very small fix to reading tif files with zero description_offset. The issue is explained in #243.

Resolves #243
closed 2021-05-31 by JacksonMaxfield