AllenCellModeling/aicsimageio #249: AICSImageIO version 4.0.0 throws error when trying to read multi-scene CZI file

sebi06, 2021-06-08:
## System and Software
* aicsimageio Version: 4.0.0 and aicspylibczi 3.0.1
* Python 3.8.10 (default, May 19 2021, 13:12:57) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.22.0 -- An enhanced Interactive Python. Type '?' for help.
PyDev console: using IPython 7.22.0
Python 3.8.10 (default, May 19 2021, 13:12:57) [MSC v.1916 64 bit (AMD64)] on win32
* Operating System: Win10

## Description
I tried to read an mutiscene (not mosaic) CZI file (96 scenes,1 Position per scene, 2 Channels). It works well, when I use aicspylibczi directly and returns the correct dimensions etc. but give the following error

Testfile: https://www.dropbox.com/s/j1vbwce0r30pmfo/testwell96.czi?dl=0

Every scene contains just a single position with X=1960 and Y=1416, but AICSImageIO assumes X should be = 1961.

```
CZI is Mosaic : False
CZI is RGB : False
No Z Dimension : 'SizeZ'
No T Dimension : 'SizeT'
No H Dimension : 'SizeH'
No I Dimension : 'SizeI'
No V Dimension : 'SizeV'
Error extracting Z Scale  : list index out of range
No Objective Name : 'Name'
No Objective ID : 'Id'
DetectorID not found : 'Id'
DetectorName not found : 'Name'
DetectorType not found : 'Type'
Trying to extract Scene and Well information if existing ...
---------------------------------------------------------
Number of Scenes :  96
0 1960 1416 168 0
1 1960 1416 19926 24
2 1960 1416 39684 49
3 1960 1416 59442 73
4 1960 1416 79200 97
5 1960 1416 98958 121
6 1960 1416 118716 145
7 1960 1416 138474 169
8 1960 1416 158232 193
9 1960 1416 177990 218
10 1960 1416 197748 242
11 1960 1416 217506 266
12 1960 1416 217482 19977
13 1960 1416 197724 19953
14 1960 1416 177966 19929
15 1960 1416 158208 19905
16 1960 1416 138450 19881
17 1960 1416 118692 19857
18 1960 1416 98934 19832
19 1960 1416 79176 19808
20 1960 1416 59418 19784
21 1960 1416 39660 19760
22 1960 1416 19902 19736
23 1960 1416 144 19712
24 1960 1416 120 39423
25 1960 1416 19878 39447
26 1960 1416 39636 39472
27 1960 1416 59394 39496
28 1960 1416 79152 39520
29 1960 1416 98910 39544
30 1960 1416 118668 39568
31 1960 1416 138426 39592
32 1960 1416 158184 39616
33 1960 1416 177942 39641
34 1960 1416 197700 39665
35 1960 1416 217458 39689
36 1960 1416 217434 59400
[... 369 lines cut ...]
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-53b885f447be>", line 1, in <module>
    runfile('C:/Users/m1srh/Documents/GitHub/czifiletools/test_metadata_aicspylibczi.py', wdir='C:/Users/m1srh/Documents/GitHub/czifiletools')
  File "C:\Program Files\JetBrains\PyCharm 2021.1.1\plugins\python\helpers\pydev\_pydev_bundle\pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm 2021.1.1\plugins\python\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:/Users/m1srh/Documents/GitHub/czifiletools/test_metadata_aicspylibczi.py", line 70, in <module>
    print(aics_img.shape)
  File "C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\aics_image.py", line 489, in shape
    return self.xarray_dask_data.shape
  File "C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\aics_image.py", line 365, in xarray_dask_data
    and dimensions.DimensionNames.MosaicTile in self.reader.dims.order
  File "C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\readers\reader.py", line 483, in dims
    self._dims = Dimensions(dims=self.xarray_dask_data.dims, shape=self.shape)
  File "C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\readers\reader.py", line 310, in xarray_dask_data
    self._xarray_dask_data = self._read_delayed()
  File "C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\aicsimageio\readers\czi_reader.py", line 517, in _read_delayed
    return xr.DataArray(
  File "C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\xarray\core\dataarray.py", line 403, in __init__
    coords, dims = _infer_coords_and_dims(data.shape, coords, dims)
  File "C:\ProgramData\Anaconda3\envs\imageanalysis_czi\lib\site-packages\xarray\core\dataarray.py", line 152, in _infer_coords_and_dims
    raise ValueError(
ValueError: conflicting sizes for dimension 'X': length 1960 on the data but length 1961 on coordinate 'X'
```

JacksonMaxfield, 2021-06-08:
Found the bug. It was floating point math. :upside_down_face: 

Patch soon!

sebi06, 2021-06-08:
Hi @JacksonMaxfield,

Isn't it mostly floating point tricks 😉.
I will do some more testing soon using my collection of nasty CZI files.

toloudis, 2021-06-08:
> Hi @JacksonMaxfield,
> 
> Isn't it mostly floating point tricks 😉.
> I will do some more testing soon using my collection of nasty CZI files.

This kind of testing is super valuable and much appreciated!

closed 2021-06-08 by JacksonMaxfield

JacksonMaxfield, 2021-06-08:
@sebi06 4.0.1 released to fix this.

sebi06, 2021-06-08:
Well since we are the ones responsible for creating the CZI with our ZEN software and since I am the PO for our data format it is part of my job (kind of)

JacksonMaxfield, 2021-06-08:
If you have any files under <400 MB that are unruley and wanted to add tests to this repo I think we would be more than welcome to accept just all the various test cases you can come up with haha. It's generally hard to add tests for stuff like this because a lot of data isn't "open access" so if you have weird data that is, feel free to send it over and add a PR :)

PR #250 (2021-06-08): bugfix/coords-and-floating-point-math-and-czi-scene-naming
## Description

Resolving all issues found in the file provided in #249 

1. The floating point math in `np.arange`, basically, across the board it is "safer" to create the array with integers and then multiply by the physical pixel size rather than creating the coord array with the start and stop as floats.
2. The provided file was a good test case for 96 scenes (or just "lots" of scenes) and when I tried to test I found that the scene names were all the same.

## Pull request recommendations:
- [x] Name your pull request _your-develo [... cut]