AllenCellModeling/aicsimageio #251: Corrupted few columns of CZI file with 4.0.0

boazmohar, 2021-06-08:
## System and Software
* aicsimageio Version: 4.0.0
* Python Version: 3.8.10
* Operating System: linux [Oracle Linux Server release 8.3]

## Description
Corrupted few columns of czi data.

## Expected Behavior
Loading it with CziFile or native software shows there is no corruption in the raw data

## Reproduction
```python 
import aicsimageio
print(aicsimageio.__version__)
>>4.0.0
from aicsimageio import AICSImage
img = AICSImage('/nrs/svoboda/moharb/ExM/20210606_B6_IHC1/IHC1/IHC1.czi', chunk_by_dims=['CZ'])
print(img.shape)
>>(1, 4, 2170, 1920, 1920)
dask_array = img.dask_data
```
[image]

```python 
plt.imshow(xarray[0,0,1,:,:], vmax=1000)
```
[image]

## Environment
 aicspylibczi-3.0.1

I can upload the file to dropbox but it is 64Gb (lmk).
Thanks!

JacksonMaxfield, 2021-06-08:
Hey! Thanks for the bug report! Hopefully we can fix it soon.

By `CziFile` you mean Gohlke's CziFile library? (we have a CziFile object in our reader which is why I need the clarification)

And yes please share the file if you can.

JacksonMaxfield, 2021-06-08:
Also, just wanted to say that I see you are using `chunk_by_dims`, we changed that to just `chunk_dims` in 4.0.

JacksonMaxfield, 2021-06-08:
Ignore my last statement. I see that `CziReader` still uses `chunk_by_dims`

boazmohar, 2021-06-08:
Hi,

1. Yes this is the [library](https://pypi.org/project/czifile/)
2. I found this still happens with a smaller [file](https://www.dropbox.com/s/1q51smagvtexzi4/Overview.czi?dl=0).

I think it is looping the right side of the image to the left side.
These were captured on a recently updated Z1 light-sheet, if that helps.
[image]

JacksonMaxfield, 2021-06-08:
Thanks for the smaller test file. We have had problems with light sheet images before... https://github.com/AllenCellModeling/aicspylibczi/issues/81

I will try to put some time at this tonight or Thursday night.

boazmohar, 2021-06-08:
@JacksonMaxfield Thanks! 
I read up on the issue you linked and I wanted to clarify that my data doesn't have "raw" and "deskewed" as it is not from the Lattice microscope (I think I only have "raw").

JacksonMaxfield, 2021-06-09:
Small update. Seems this issue is down at the `aicspylibczi` level (our library for reading CZI files) which means it's likely somewhere in the C++ code.

boazmohar, 2021-06-09:
Hi @JacksonMaxfield, thanks for the update. Would you like me to open an issue there?

Boaz

JacksonMaxfield, 2021-06-09:
Nope no worries. I will simply transfer this issue over to that repo if needed but leaving it here until next dev meeting which will be on Monday.

JacksonMaxfield, 2021-06-16:
A couple of updates.

1. I am going to transfer this issue to the aicspylibczi repo as it seems to be an issue on that side.
2. I personally don't have a timeline for resolution but I think @toloudis may have time to look at it? The general statement is, we will keep tracking this issue and post updates.

Sorry for the long process.