AllenCellModeling/aicsimageio #252: How to open only a tile/part of a .czi file

LRobach, 2021-06-08:
## Use Case
*Please provide a use case to help us understand your request in context*
Hello ! 
I have to open a .czi file but because it is too heavy, I would like to open only a tile of it. 
I could not found anything about .dask_data, get_image_data and get_image_dask_data regarding to .czi files… I do not know if I could use them for these kind of output.

JacksonMaxfield, 2021-06-08:
Hey @LRobach!

First, a clarifying question: Is your file natively tiled? When you aquired the image was it aquired as a mosaic?

In regards to `.dask_data`, `get_image_data`, and `get_image_dask_data` all of these functions work for you file. All functions and properties work for all the supported file types of the library. (With a few very rare exceptions, and in which case you will get a `NotImplementedError` usually)

If your file is too large for memory, use `dask_data` or `get_image_dask_data` and `compute()` the chunk you want.
If your file fits in memory, use `data` or `get_image_data`.

JacksonMaxfield, 2021-06-14:
Just wanted to check and see if you have any other questions. Can I close this?

LRobach, 2021-06-15:
Hello ! 
Sorry I did not answer back, in fact your answer helps me a lot and I forgot to answer you back !
Thanks a lot

JacksonMaxfield, 2021-06-15:
Great! Glad to help.

closed 2021-06-15 by JacksonMaxfield