AllenCellModeling/aicsimageio #254: Lazy Reading - AICSImageIO is always returning 5D stacks from 6D stack CZI

sebi06, 2021-06-09:
Hi,

this is a usage question.

I have a CZI file with 96 scenes (6D Stack). It is clear how to read a single scene also using the "lazy" option. But since AICSImageIO always return a 5D stack without the scene dimension I am wondering what the best approach is to read all scenes as once using a delayed reader.

- Do I have to create my own dask array with 96 addition chunks and pass it on to Napari etc.?
- Or is the a easier (smarter) built-in way which I did not get yet?

JacksonMaxfield, 2021-06-09:
Hey @sebi06 

Yea there is no built in way to get the whole stack in one go. I think we probably could add one.... :thinking: 
I will talk with @toloudis about this next dev meeting.

For now best you can do is:
```python
from aicsimageio import AICSImage
import dask.array as da

img = AICSImage("testwell96.czi")
stacks = []
for scene in img.scenes:
    img.set_scene(scene)
    stacks.append(img.dask_data)

stacks = da.stack(stacks)
```

Will let you know what we talk about during dev meeting on this topic.

toloudis, 2021-06-09:
we might be able to just wrap the code snippet you suggested, in a function.  But we still have that nasty problem of what if the scene names are not unique.

sebi06, 2021-06-09:
The code snippet is fine for now. No urgent need to add it. Let's see how people use what is there.

JacksonMaxfield, 2021-06-16:
Closing. Tracking the `set_scene_from_index` in #259

closed 2021-06-16 by JacksonMaxfield