AllenCellModeling/aicsimageio #255: Reconstruct Stiched Images does not work for CZI

sebi06, 2021-06-09:
## System and Software
* aicsimageio Version: 4.0.1
* Python Version: 3.8.10
* Operating System: Win10

## Description

In your twitter post I saw this:

[image]

https://www.dropbox.com/s/i3t8vi7f4vypsg3/OverviewScan.czi?dl=0

But when I try this on a CZI that has 120 Tile I get using

```python
aics_img = AICSImage(filename)
print(aics_img.shape)
for k,v in aics_img.dims.items():
    print(k,v)
```

```
(120, 1, 1, 1, 440, 544)
M 120
T 1
C 1
Z 1
Y 440
X 544
```

JacksonMaxfield, 2021-06-09:
Uhhhhhh fascinating.... I can't reproduce.

```python
In [1]: from aicsimageio import AICSImage

In [2]: img = AICSImage("OverviewScan.czi")

In [3]: img.dims
Out[3]: <Dimensions [T: 1, C: 1, Z: 1, Y: 3212, X: 7398]>
```

And, funny enough. I think this is the exact file we test against.... https://github.com/AllenCellModeling/aicsimageio/blob/main/aicsimageio/tests/test_aics_image.py#L964

We test on Windows so it doesn't seem like a windows thing.

I am truly confused.

Would you mind pasting the output of `pip freeze`

sebi06, 2021-06-09:
Very interesting indeed. I noticed that in you docu only the LIF reader is mentioned to support stitching

https://github.com/AllenCellModeling/aicsimageio#mosaic-image-reading

Here is the output:

```
Python 3.8.10 (default, May 19 2021, 13:12:57) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.22.0 -- An enhanced Interactive Python. Type '?' for help.
PyDev console: using IPython 7.22.0
Python 3.8.10 (default, May 19 2021, 13:12:57) [MSC v.1916 64 bit (AMD64)] on win32
pip freeze
aicsimageio==4.0.1
aicspylibczi==3.0.1
alabaster==0.7.12
apeer-dev-kit==1.2.2
apeer-ometiff-library==1.7.2
appdirs==1.4.4
argon2-cffi @ file:///C:/ci/argon2-cffi_1613037959010/work
asciitree==0.3.3
async-generator @ file:///home/ktietz/src/ci/async_generator_1611927993394/work
atomicwrites==1.4.0
attrs @ file:///tmp/build/80754af9/attrs_1620827162558/work
autopep8 @ file:///tmp/build/80754af9/autopep8_1620866417880/work
Babel==2.9.1
backcall @ file:///home/ktietz/src/ci/backcall_1611930011877/work
bleach @ file:///tmp/build/80754af9/bleach_1612211392645/work
bokeh @ file:///C:/ci/bokeh_1620784067744/work
brotlipy==0.7.0
cachey==0.2.1
certifi==2021.5.30
cffi @ file:///C:/ci/cffi_1613247279197/work
chardet @ file:///C:/ci/chardet_1607690654534/work
click @ file:///tmp/build/80754af9/click_1621604852318/work
cloudpickle @ file:///tmp/build/80754af9/cloudpickle_1598884132938/work
colorama @ file:///tmp/build/80754af9/colorama_1607707115595/work
conda==4.10.1
conda-package-handling @ file:///C:/ci/conda-package-handling_1618262320430/work
cryptography @ file:///C:/ci/cryptography_1616769344312/work
cycler==0.10.0
cytoolz==0.11.0
dask==2021.6.0
dask-image @ file:///tmp/build/80754af9/dask-image_1599011548464/work
decorator @ file:///tmp/build/80754af9/decorator_1621259047763/work
defusedxml @ file:///tmp/build/80754af9/defusedxml_1615228127516/work
distributed @ file:///C:/ci/distributed_1617384289923/work
dnspython==2.1.0
docstring-parser==0.7.3
docutils @ file:///C:/ci/docutils_1620828260973/work
elementpath==2.2.2
email-validator==1.1.2
entrypoints==0.3
et-xmlfile==1.1.0
fasteners @ file:///tmp/build/80754af9/fasteners_1608142287400/work
freetype-py==2.2.0
fsspec==2021.6.0
HeapDict==1.0.1
idna @ file:///home/linux1/recipes/ci/idna_1610986105248/work
imagecodecs @ file:///C:/ci/imagecodecs_1621408025737/work
imageio @ file:///tmp/build/80754af9/imageio_1617700267927/work
imageio-ffmpeg==0.4.4
imagesize==1.2.0
importlib-metadata @ file:///C:/ci/importlib-metadata_1617877484576/work
iniconfig @ file:///home/linux1/recipes/ci/iniconfig_1610983019677/work
ipykernel @ file:///C:/ci/ipykernel_1596190155316/work/dist/ipykernel-5.3.4-py3-none-any.whl
ipython @ file:///C:/ci/ipython_1617121002983/work
ipython-genutils @ file:///tmp/build/80754af9/ipython_genutils_1606773439826/work
ipywidgets @ file:///tmp/build/80754af9/ipywidgets_1610481889018/work
jdcal==1.4.1
[... 114 lines cut ...]
testpath @ file:///home/ktietz/src/ci/testpath_1611930608132/work
tifffile @ file:///tmp/build/80754af9/tifffile_1619636090847/work
toml @ file:///tmp/build/80754af9/toml_1616166611790/work
toolz @ file:///home/linux1/recipes/ci/toolz_1610987900194/work
tornado @ file:///C:/ci/tornado_1606942392901/work
tqdm @ file:///tmp/build/80754af9/tqdm_1615925068909/work
traitlets @ file:///home/ktietz/src/ci/traitlets_1611929699868/work
transforms3d==0.3.1
typing-extensions @ file:///home/ktietz/src/ci_mi/typing_extensions_1612808209620/work
urllib3 @ file:///tmp/build/80754af9/urllib3_1615837158687/work
vispy==0.6.6
wcwidth @ file:///tmp/build/80754af9/wcwidth_1593447189090/work
webencodings==0.5.1
widgetsnbextension==3.5.1
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306167264/work
win32-setctime==1.0.3
wincertstore==0.2
wrapt==1.12.1
xarray==0.16.2
xmlschema==1.4.1
xmltodict==0.12.0
zarr==2.6.1
zict==2.0.0
zipp @ file:///tmp/build/80754af9/zipp_1615904174917/work
Note: you may need to restart the kernel to use updated packages.

JacksonMaxfield, 2021-06-09:
Woops. The docs are simply just outdated. CZI is definitely supported.

Could you make a brand new env?

```bash
conda create --name new-test-env python=3.9 -y
conda activate new-test-env
pip install aicsimageio
```

sebi06, 2021-06-09:
You probably mean pip install aivsimageio[all] etc. because otherwise it will not read CZI at all, right?

JacksonMaxfield, 2021-06-09:
Just `[czi]` should be enough

sebi06, 2021-06-09:
Ok, i tried it and now it works ...

```
(1, 1, 1, 3212, 7398)
T 1
C 1
Z 1
Y 3212
X 7398
```

And in that case pip freeze give me:

```
(new_test_env) C:\Users\m1srh\Documents\GitHub\czifiletools>pip freeze
aicsimageio==4.0.1
aicspylibczi==3.0.1
asciitree==0.3.3
certifi==2021.5.30
cloudpickle==1.6.0
dask==2021.6.0
dnspython==2.1.0
elementpath==2.2.2
email-validator==1.1.2
fasteners==0.16.2
fsspec==2021.6.0
idna==3.2
imagecodecs==2021.6.8
imageio==2.9.0
imageio-ffmpeg==0.4.4
locket==0.2.1
numcodecs==0.7.3
numpy==1.20.3
ome-types==0.2.4
packaging==20.9
pandas==1.2.4
partd==1.2.0
Pillow==8.2.0
Pint==0.17
pydantic==1.8.2
pyparsing==2.4.7
python-dateutil==2.8.1
pytz==2021.1
PyYAML==5.4.1
readlif==0.6.1
six==1.16.0
tifffile==2021.6.6
toolz==0.11.1
typing-extensions==3.10.0.0
wincertstore==0.2
xarray==0.16.2
xmlschema==1.4.1
zarr==2.6.1
```

JacksonMaxfield, 2021-06-09:
Weird. I wonder which part of the env was corrupted :shrug:

closed 2021-06-09 by JacksonMaxfield

sebi06, 2021-06-09:
OK, so it works in principle, which is good.
Not so good, my productive environment with my other required libs is not working...

JacksonMaxfield, 2021-06-09:
You could try doing a:

`pip install --force-reinstall --no-cache-dir aicsimageio[czi]==4.0.1` in your production env

sebi06, 2021-06-09:
Now it works as well.

Python dependencies are always fun. You can imagine the fun we have, when releasin our ZeissPy 4times a year while trying to stay compatible with all requirements from all projects in microscopy (at ZEISS).

Thnaks for you help.