AllenCellModeling/aicsimageio #259: Add `set_scene_from_index` function

JacksonMaxfield, 2021-06-16:
## Use Case
*Please provide a use case to help us understand your request in context*

We have operated under the assumption that scene names are unique identifiers and so only have a `set_scene` function but we don't actually do any checking on the scene names for uniqueness so for future proofing best if we simply add this function.

## Solution
*Please describe your ideal solution*

function def probably like:
```python
def set_scene_from_index(index: int) -> None:
    """
    Set current operating scene with integer index.

    Parameters
    ----------
    index: int
        The scene index to change to.
    """
```

## Alternatives
*Please describe any alternatives you've considered, even if you've dismissed them*

smzgit, 2021-06-18:
Hi, Can you explain what is current operating scene ?

JacksonMaxfield, 2021-06-18:
> Hi, Can you explain what is current operating scene ?

Many microscopy image formats have a concept of "scenes" or "sub-images", these scenes or sub-images can have totally different dimensions, data types (uint8, vs uint16, etc), and more. This function updates which scene or sub-image the AICSImage and Reader object is operating against.

If you are just learning, that's really all there is. Check out our README for more info.

If you are looking to contribute, I updated this issue a tiny bit, and you should look at the current existing function `set_scene` and simply add the functionality to it.

PR #272 (2021-07-07): feature/allow-set-scene-by-index
## Description

Allows setting scene by index. The required a small bit of extra changes where we now store current scene index instead of current scene as the basis for scene management.

## Pull request recommendations:
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-12], adds tiff file format support_

Resolves #259 

- [x] Provide relevant tests for your feature or bug fix.
- [x] Provide or update documentation for any feature adde [... cut]
closed 2021-07-07 by JacksonMaxfield