AllenCellModeling/aicsimageio #262: function to convert Zeiss .czi files to DeepZoom .dzi using aicsimageio and pyvips

rcasero, 2021-06-23:
Hi,

I've written a little function to convert Zeiss .czi files to DeepZoom .dzi using aicsimageio and pyvips (in my case, my interest is to be able to visualise them in the web browser application [AIDA](https://github.com/alanaberdeen/AIDA)).

https://github.com/MRC-Harwell/cytometer/blob/c96e86c144c63978881330ad8275d10649096fcd/cytometer/data.py#L1525

The functionality is a bit limited to the Zeiss files I'm using now, and there may be a better way to do this, but I thought I'd drop a link here, as maybe somebody else is interested.

JacksonMaxfield, 2021-06-23:
Hey @rcasero this is awesome!

We may make an "AICSImageIO Examples" section in documentation and link to this if that is okay! No timeline on that at all just a thought.

Three sidenotes:
1. `AICSImage.dims` doesn't load any of the image data into memory (or at least it shouldn't...) so you should be able to get the Y and X from the `dims` object too instead of parsing metadata.
2. If the aim is a perfect conversion from one to another you may want to use `AICSImage.reader.data` to get all the data and dimensions in the file as the `AICSImage` object will only pull the "OME standard dimensions".
3. Should we add a `DziReader` to the library?

toloudis, 2021-06-23:
Interesting side note: I used dzi format (in conjunction with lots of aicsimageio processing) to make this visualization with OpenSeaDragon:  https://www.allencell.org/deep-cell-zoom.html

rcasero, 2021-07-06:
> 1. `AICSImage.dims` doesn't load any of the image data into memory (or at least it shouldn't...) so you should be able to get the Y and X from the `dims` object too instead of parsing metadata.

@JacksonMaxfield That's what I thought, but I have an image with `<Dimensions [T: 1, C: 1, Z: 1, Y: 76766, X: 131243, S: 3]>`, and doing `im.dims` takes a good 10 min or longer. I'll open a new issue for that.

rcasero, 2021-07-06:
> 2. If the aim is a perfect conversion from one to another you may want to use `AICSImage.reader.data` to get all the data and dimensions in the file as the `AICSImage` object will only pull the "OME standard dimensions".

So far, for my project, I think the standard dimensions work, but I'm happy if somebody wants to improve or reuse my code  under the Apache licence.

closed 2023-04-13 by github-actions[bot]