AllenCellModeling/aicsimageio #263: aicsimagio fails to read tiff file while imageio suceeds

donovanr, 2021-06-25:
## System and Software
* aicsimageio Version: 4.0.2
* Python Version: 3.8.0
* Operating System: Ubuntu 18.04.5 LTS (Bionic Beaver)

## Description
Using `AICSImage` to load up a single channel tiff gives an error when trying to access the data.  `imageio.imread` loads it up ok.

## Reproduction
In the following, `p` is a `pathlib.Path` to the image file, which is a single channel .TIFF (_not_ ome, no metadata).

Using plain imageio works:
```python
imageio_img = imageio.imread(p)
imageio_img.shape
(1104, 1104)
```

This doesn't:
```python
aics_img = AICSImage(p, reader=TiffReader)
aics_img.data
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-19-1508f5ddc75d> in <module>
      1 aicsimg = AICSImage(p, reader=TiffReader)
----> 2 aicsimg.data

~/miniconda3/envs/dsci/lib/python3.8/site-packages/aicsimageio/aics_image.py in data(self)
    467         Recommended to use `dask_data` for large mosaic images.
    468         """
--> 469         return self.xarray_data.data
    470 
    471     @property

~/miniconda3/envs/dsci/lib/python3.8/site-packages/aicsimageio/aics_image.py in xarray_data(self)
    407                 self._reconstruct_mosaic
    408                 # Does the data have a tile dim
--> 409                 and dimensions.DimensionNames.MosaicTile in self.reader.dims.order
    410             ):
    411                 try:

~/miniconda3/envs/dsci/lib/python3.8/site-packages/aicsimageio/readers/reader.py in dims(self)
    481         """
    482         if self._dims is None:
--> 483             self._dims = Dimensions(dims=self.xarray_dask_data.dims, shape=self.shape)
    484 
    485         return self._dims

~/miniconda3/envs/dsci/lib/python3.8/site-packages/aicsimageio/readers/reader.py in xarray_dask_data(self)
    308         """
    309         if self._xarray_dask_data is None:
--> 310             self._xarray_dask_data = self._read_delayed()
    311 
    312         return self._xarray_dask_data

~/miniconda3/envs/dsci/lib/python3.8/site-packages/aicsimageio/readers/tiff_reader.py in _read_delayed(self)
    451                     attrs={
    452                         constants.METADATA_UNPROCESSED: tiff_tags,
--> 453                         constants.METADATA_PROCESSED: tiff_tags[
    454                             TIFF_IMAGE_DESCRIPTION_TAG_INDEX
    455                         ].value,

~/miniconda3/envs/dsci/lib/python3.8/site-packages/tifffile/tifffile.py in __getitem__(self, key)
   7925             return self._dict[key]
   7926         if not isinstance(key, str):
-> 7927             raise KeyError(key)
   7928         for tag in self._dict.values():
   7929             if tag.name == key:

KeyError: 270
```

Calling `AICSImage` without the reader kwarg gives an immediate error before even trying to acess the data:
``` python
aicsimg = AICSImage(p)
[ERROR: ome_tiff_reader:  98 2021-06-25 21:25:57,933] Failed to parse XML for the provided file.
[ERROR: ome_tiff_reader:  99 2021-06-25 21:25:57,934] no element found: line 1, column 0
```

## Environment
I'm working inside a conda env I made for this project on a remote machine, nothing unusual.

JacksonMaxfield, 2021-06-25:
Looks like a metadata parsing bug.... This image doesn't have a description???

Can you send me the file @donovanr. Privately is fine: jmaxfieldbrown@gmail.com

toloudis, 2021-06-26:
Sounds like we just need to handle that case where there is no ImageDescription tag?

JacksonMaxfield, 2021-06-26:
Yes. I still would like @donovanr to send over an example of the file. It's pretty weird imo to have a TIFF with a missing tag and I want to see if anything else fails tests with it.

jo-mueller, 2021-06-29:
I think I am having a similar issue with tif images (aicsimageio version = 3.3.4). The error I am getting is rather uninformative:

```
img = AICSImage(filename)
*** TypeError: string argument without an encoding
```
while importing the data with tifffile works fine:
```
img = tf.imread(filename)
```

Would you mind if I send over an example image as well?

JacksonMaxfield, 2021-06-29:
> I think I am having a similar issue with tif images (aicsimageio version = 3.3.4). The error I am getting is rather uninformative:
>
> ```
> img = AICSImage(filename)
> *** TypeError: string argument [...]

Hey @jo-mueller your issue should be entirely different unfortunately. But should be much easier to resolve. Can you provide a longer snippet of code that is causing the issue. Specifically how are you setting up the filename variable. The only reason that `TypeError` should occur is if you are providing it as a bytes object.... AICSImageIO v3 (and v4) has some additional path handling on top of base tifffile.

Also you may just want to try upgrading to 4.0.2 and see if that resolves your issue.

closed 2021-07-05 by JacksonMaxfield

JacksonMaxfield, 2021-07-05:
This will be resolved in 4.0.3 which is currently going through CI: https://github.com/AllenCellModeling/aicsimageio/runs/2992265505

As soon as all tests are done, a "publish" job will run.

PR #271 (2021-07-05): bugfix/missing-tiff-description-tag
## Description

@donovanr's TIFFs are weird and don't have a TIFF description tag (270), so this is to catch that possibility.

## Pull request recommendations:
- [x] Name your pull request _your-development-type/short-description_. Ex: _feature/read-tiff-files_
- [x] Link to any relevant issue in the PR description. Ex: _Resolves [gh-12], adds tiff file format support_

Resolves #263 

- [ ] Provide relevant tests for your feature or bug fix.

Data is private but mypy succeeds so should be okay! It's nice to know mypy can act as a backup when data is privat [... cut]