AllenCellModeling/aicsimageio #264: OmeTiffWriter.save with channel_names=List[str] fails

donovanr, 2021-06-25:
## System and Software
* aicsimageio Version: 4.0.2
* Python Version: 3.8.0
* Operating System: Ubuntu 18.04.5 LTS (Bionic Beaver)

## Description
`OmeTiffWriter.save` fails / gives an error when asked to use a list of strings as channels names.  Using a list of a list of strings works, but is unintuitive that this should be the expected behavior.

## Expected Behavior
A list of strings for a single scene image should be acceptable input to `channel_names`

## Reproduction
using `List[List[str]]` for channel names works:
```python
image0 = np.ndarray([1, 3, 1, 256, 256])
OmeTiffWriter.save(
    image0,
    "file.ome.tif",
    dim_order="TCZYX",
    channel_names=[["C00","C01","C02"]],
)
```

While just `List[str]` gives an error:

```python
image0 = np.ndarray([1, 3, 1, 256, 256])
OmeTiffWriter.save(
    image0,
    "file.ome.tif",
    dim_order="TCZYX",
    channel_names=["C00","C01","C02"],
)

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-93-52f06d12283d> in <module>
      1 image0 = np.ndarray([1, 3, 1, 256, 256])
----> 2 OmeTiffWriter.save(
      3     image0,
      4     "file.ome.tif",
      5     dim_order="TCZYX",

~/miniconda3/envs/dsci/lib/python3.8/site-packages/aicsimageio/writers/ome_tiff_writer.py in save(data, uri, dim_order, ome_xml, channel_names, image_name, physical_pixel_sizes, channel_colors, **kwargs)
    212         # try to construct OME from params
    213         if ome_xml is None:
--> 214             ome_xml = OmeTiffWriter.build_ome(
    215                 [i.shape for i in data],
    216                 [i.dtype for i in data],

~/miniconda3/envs/dsci/lib/python3.8/site-packages/aicsimageio/writers/ome_tiff_writer.py in build_ome(data_shapes, data_types, dimension_order, channel_names, image_name, physical_pixel_sizes, channel_colors)
    582             or num_images != len(channel_colors)
    583         ):
--> 584             raise ValueError("Mismatched array counts in parameters to build_ome")
    585 
    586         images = []

ValueError: Mismatched array counts in parameters to build_ome
```

I would expect a normal list to work for single scene images.

JacksonMaxfield, 2021-06-25:
@toloudis can you check this? I think it's because the array being provided is a list of array and the channel names is a list of lists but I thought that is correct...

JacksonMaxfield, 2021-06-25:
@donovanr you may just want to provide the single numpy array and just a single list of string and see if that works.

JacksonMaxfield, 2021-06-26:
Oh I'm bad. You already had this figured out. Sorry the reason this is happening is because when providing a `List[ArrayLike]` the assumption is that you are providing multiple scenes and so you must then provide a `List[List[str]]` for channel names and other metadata, one sublist per scene.

Any ideas as to how to better document this?

toloudis, 2021-06-26:
I think you stated it pretty well: if you pass a list of 5D arrays, then it expects a list of lists of channel names.  Where the outermost list is the set of scenes.    But it seems to me that the given case in the bug should be working, no?

toloudis, 2021-06-29:
Thank you for the bug report. Confirming this was definitely a bug. Fix will be coming very soon.

PR #265 (2021-06-30): fix channel_names array expansion test
## Description
Fix #264
Channel names were testing against int, which prevented the array from being expanded properly.
closed 2021-06-30 by JacksonMaxfield