AllenCellModeling/aicsimageio #266: Installing aicsimageio with python 3.7

rcasero, 2021-07-01:
## Use Case

I need to run `aicsimageio` with older software that have packages up to python 3.7. However, aicsimageio installed with 
```
pip install "aicsimageio[czi] @ git+https://github.com/AllenCellModeling/aicsimageio.git"
```
requires python 3.8 or 3.9.

## Solution

Ideally, if `setup.py` can be modified to add to `setup()` the line

```
        "Programming Language :: Python :: 3.7",
```

and that works, that'd solve the incompatibility error. However, I'm not sure whether that'd break something. 

## Alternatives

I could fork the project, modify it, and then if it works, use my fork. However, then it needs to be kept in sync with the original project. That's why I thought about asking first.

JacksonMaxfield, 2021-07-01:
Hey @rcasero! This makes total sense. And to be honest.... I think this was entirely my bad. I thought Py37 support was dropped from `tifffile` which is a pretty core dependency but it looks like I was wrong in that thought. I just tried installing and the latest `tifffile` works just fine so don't know where I got that idea.

It's a bit more involved than just that classifier iirc. For best support it would be great to add that classifier and:
* [ ] [change the `python_requires`](https://github.com/AllenCellModeling/aicsimageio/blob/main/setup.py#L110)
* [ ] [add py37 to tox for testing](https://github.com/AllenCellModeling/aicsimageio/blob/main/tox.ini#L3)
* [ ] [add 3.7 to github actions for build main](https://github.com/AllenCellModeling/aicsimageio/blob/main/.github/workflows/build-main.yml#L19)
* [ ] [add 3.7 to github actions for PRs](https://github.com/AllenCellModeling/aicsimageio/blob/main/.github/workflows/test-and-lint.yml#L11)

If you are game to make all those changes, I would gladly accept PR. I can also make those thise weekend.

rcasero, 2021-07-02:
Hi @JacksonMaxfield ,

Thanks for the reply. I'll give it a try!

rcasero, 2021-07-04:
I have create a PR (https://github.com/AllenCellModeling/aicsimageio/pull/268) with the suggested changes, although as noted in  issue #267, there were `make build` errors even before making any changes.

JacksonMaxfield, 2021-07-04:
Huh. Sorry for all the trouble in this @rcasero, I know it was a struggle but I really appreciate it. It forces me to understand the "startup cost" of setting up the project.

It looks like most of the errors you were seeing in that log were simply because you don't have any AWS credentials on that machine. Which I will make a note of in an issue. Contributing docs should really say:

* If you have any default AWS credentials, run `make build` to include remote IO tests
* If you do not have any AWS credentials, run `make build-local` to only run local IO tests

I will have a PR later this weekend to hopefully resolve some of the startup issues you encountered... I really do appreciate you pointing them out and making a PR for this!

closed 2021-07-05 by JacksonMaxfield

JacksonMaxfield, 2021-07-05:
New version will be published in ~1 hour

toloudis, 2021-07-05:
> 
> * If you have any default AWS credentials, run `make build` to include remote IO tests
> * If you do not have any AWS credentials, run `make build-local` to only run local IO tests
> 
Maybe build-local should be the default?  
* If you have any default AWS credentials, run `make build-with-remote` to include remote IO tests
* If you do not have any AWS credentials, run `make build` to only run local IO tests

JacksonMaxfield, 2021-07-05:
Yep I was thinking the same thing.

On the CI side, PRs would run local tests only and build-main would run full suite. We would catch regressions late but it should also be pretty obvious if someone is making buffer handling changes in code.

PR #270 (2021-07-05): @rcasero-feature/add-python-3.7-support
## Description

See #269 and #268 for prior PRs during the process.

This adds Python 3.7 support back to the library. Also during the process we found out that a new `Pillow` version was released that broke a lot of builds (including our if we didn't patch this today probably). So I changed how we can specify deps for a format, instead of just specifying one dep for a format, you can now provide a list of deps and it will be unpacked. In this case, to specifically ignore a `Pillow` version for `DefaultReader` / `TimeseriesWriter`.

## Pull request recom [... cut]